arduino/Arduino #4369: Request to add virtmem to the library manager

rhelmus, 2016-01-01:
Hi all,

I would like to request the addition of my first release of the virtmem library.

Arduino files: https://github.com/rhelmus/virtmem/tree/master/virtmem
Complete repos: https://github.com/rhelmus/virtmem

Only the files in the (former linked) subdirectory are necessary for Arduino and should follow the new spec. I hope it is not a problem for the library manager that this directory is not in the root?

Thanks!

cmaglie, 2016-01-04:
Hi @rhelmus,
the publishing process requires that the library must be on the root folder of the repository, so you must move the content of:

https://github.com/rhelmus/virtmem/virtmeme/tree/v1.0

to

https://github.com/rhelmus/virtmem/tree/v1.0

I don't know if this is an option for you, I see that you have a lot of extra files and subfolders on the repository root folder. Maybe making another repo with a copy just for the release is a more feasible approach?

rhelmus, 2016-01-04:
Hi cmaglie,

Thanks for your feedback. Following your suggestion I made a special repository with only the files needed for Arduino: https://github.com/rhelmus/virtmem-arlibman/tree/v1.0

Cheers,
Rick

cmaglie, 2016-01-04:
Added, thanks!

closed 2016-01-04 by cmaglie