arduino/Arduino #4375: Arduino/app/src/processing/app/EditorStatus.java contains fragmented messages

kotobuki, 2016-01-02:
The file contains three texts to be translated as follows (line 347 to 352):

```java
if (!(PreferencesData.getBoolean("build.verbose"))) {
  message1 += "\n\n";
  message1 += "  " + tr("This report would have more information with") + "\n";
  message1 += "  \"" + tr("Show verbose output during compilation") + "\"\n";
  message1 += "  " + tr("enabled in File > Preferences.") + "\n";
}
```

Please consider a message instead of three fragmented messages, since that confuses us translators.

per1234, 2016-01-04:
https://github.com/hasso/Arduino/commit/804c15e83a390ea505aa53189fa34a49ecc9c0af in https://github.com/arduino/Arduino/pull/4381 seems to be attempting to do something like what you are requesting.

cmaglie, 2016-01-05:
fixed by #4381, I've update the translation on transifex too. Thanks!

closed 2016-01-05 by cmaglie