arduino/Arduino #4379: avrdude: can't open config file "/Users/jenkins/jenkins/workspace/toolchain-avr-mac32/objdir/etc/avrdude.conf": No such file or directory

fr0zenrain, 2016-01-03:
a problem with avrdue  In OSX 10.10 
command "avrdude -v" show errors as follows
avrdude: can't open config file "/Users/jenkins/jenkins/workspace/toolchain-avr-mac32/objdir/etc/avrdude.conf": No such file or directory
avrdude: error reading system wide configuration file "/Users/jenkins/jenkins/workspace/toolchain-avr-mac32/objdir/etc/avrdude.conf"

and i also set AVRDUDE_CONF=$AVR_TOOLS_DIR/etc/avrdude.conf 
it seems avrdue_bin use a hardcode path, i don't want create this path for configure file ,how can i resolve?

astrawa, 2016-01-04:
I have the same problem with  arduino-1.6.7-linus32 on a raspberrypi.
Previously, I got an error saying there was an "id_type" error at line 332 in avrdude.conf. I removed and re-installed arduino. Now I get the error that avrdude.conf does not exist both in the IDE and from the command line.

astrawa, 2016-01-04:
Re-installed. Now avrdude.conf can be read but I am back to the "id_type" error. 
A previous post suggested that there might be another avrdude.cnof file on the machine, but there is not.

facchinm, 2016-01-04:
Hi @fr0zenrain , the `avrdude` executable is a script (you can open it using Textedit) that wraps `avrdude_bin` executable. It doesn't wrap the `conf` path since it is passed by the IDE.
If you are in `Arduino.app/Contents/Java/hardware/tools/avr/`, the right way to execute `avrdude` is `./bin/avrdude -C etc/avrdude.conf`
@astrawa , I believe your problem is totally different. Where did you get the IDE for RPi? Can you open another issue? Thanks :wink:

closed 2016-01-04 by facchinm

chappy1978, 2017-03-09:
This is still a problem. Jenkins directory is still hard coded into avrdude

neilyoung, 2017-05-09:
Same here:

```
~/Documents/Arduino/attiny $ /Users/me/.platformio/packages/tool-avrdude/avrdude -c stk500v2 -p attiny2313a -Ulfuse:w:0x64:m
avrdude: can't open config file "/Users/jenkins/jenkins/workspace/avrdude/label/mac-mini/objdir/etc/avrdude.conf": No such file or directory
avrdude: error reading system wide configuration file "/Users/jenkins/jenkins/workspace/avrdude/label/mac-mini/objdir/etc/avrdude.conf"
```

facchinm, 2017-05-09:
The fact that `jenkins` is hardcoded as the config path is a non-problem, since you **must** specify a config file with `-C` . The Arduino IDE already performs this correctly, please contact your distributor (`platformio` or anyone else) if you are experiencing problems with their buildsystem.

Tigeruno, 2020-05-12:
avrdude wil not accept -c for programmer or -C or location of .conf file  3 days working on trying to solve this

Tigeruno, 2020-05-12:
avrdude –c avrisp –p t84 -b 19200 -P /dev/cu.usbmodem14201 –u –U flash:w:firmware.hex

avrdude: no programmer has been specified on the command line or the config file
         Specify a programmer using the -c option and try again

dhanson@Davids-pro-2039 Downloads % avrdude –c avrisp –p t84 -b 19200 -P /dev/cu.usbmodem14201 –u –U flash:w:firmware.hex

avrdude: no programmer has been specified on the command line or the config file
         Specify a programmer using the -c option and try again

Tigeruno, 2020-05-12:
Cant believe no one can solve this.

cmaglie, 2020-05-13:
> avrdude –c avrisp –p t84 -b 19200 -P /dev/cu.usbmodem14201 –u –U flash:w:firmware.hex

you are typing `–c` instead of `-c`, (a longer dash instead of the "normal" dash). I don't know where this "long dash" comes from, but if this is the command line you're using I'm pretty sure avrdude will not recognize `–c`, you must use `-c` and the same for all the other options.

Tigeruno, 2020-05-13:
Thanks for the reply, been trying to get this done for 3 long days.  This is the only dash I have on my keyboard (mac OS 10.15.4 Catalina (64bit)).  Can you give me the ascii value maybe I can do that.  Most of my work is in atom and platformIO, python 2.7.  Closest I have gotten to getting the tiny841 programmed, arduinoISP connected to DIYBMSv4 via 6 pin ISP breakout by Stuart Pittaway.  Writes but unable to verify so fails.  I originally programmed these modules several months ago using arduinoISP setup.  Only change is Mac upgrade OS.

cmaglie, 2020-05-13:
The ascii code is `45` (or `2D` in hex). You can copy and paste this one: `-` if unsure.

BTW this issue really doesn't belong this repository because is not related to Arduino or the Arduino IDE.

If you're using the `avrdude` from the Arduino IDE, as already said, the path to config file must be specified with `-C path/to/avrdude.conf`.
If you're using an `avrdude` obtained in other ways we can't help you, but probably the `-C` solution will work as well.

Said that, I'm going to lock this issue from further comments.

Tigeruno, 2020-05-13:
Thanks for that.  I will keep trying to get an answer or fix.

last response from PlatformIO:

avrdude: 376 bytes of flash written
avrdude: verifying flash memory against .pio/build/attiny841/firmware.hex:
avrdude: load data flash data from input file .pio/build/attiny841/firmware.hex:
avrdude: input file .pio/build/attiny841/firmware.hex contains 376 bytes

avrdude: reading on-chip flash data:
 
Reading | ################################################## | 100% 0.22s
 
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x00 != 0x1d
avrdude: verification error; content mismatch