arduino/Arduino #4380: Support automatic compilation and linking of files with C extension

saxbophone, 2016-01-03:
Hello! I wrote a small library in pure C, that does not call any of the Arduino library functions ( https://github.com/saxbophone/pid ) - it consists of a C header file and a C source file, along with a few other files to assist the IDE when used as a package.

However, when zipping the code up and importing into the IDE as a library, I get a linker error when verifying the code. (code compiles fine with GCC) It looks like the compiler can see my function prototype fine, but cannot see the implementation file for it. This seems to happen regardless of which board is targetted (tested and reproduced problem with Uno, Nano and 101 boards).

I am 99% sure this is because the compiler/linker is ignoring files with the `.c` extension type, as if I rename the file to have a `.cpp` extension then verifying the code works fine.

I'm aware that Arduino's codebase and the whole build process is made up of mostly C++, but it feels incorrect to change my file's extension when the code is in fact pure C with no C++ constructs used at all. 
Given C and C++ can be compiled separately and linked together (and most C code would also compile with a C++ compiler), could support for files with C extension type be added?

saxbophone, 2016-01-03:
Oh, btw - this was tested with Arduino 1.6.7 running on Ubuntu 15.10

NicoHood, 2016-01-03:
Maybe this helps?
https://github.com/arduino/Arduino/issues/4333

saxbophone, 2016-01-03:
Aha! Thank you, that did the trick! Boy do I feel foolish now :blush:

closed 2016-01-03 by saxbophone