arduino/Arduino #4382: Add mfrc522 library

Rotzbua, 2016-01-04:
Please add RFID library:
https://github.com/miguelbalboa/rfid

cmaglie, 2016-01-04:
@Rotzbua 
the release tagged 1.1.2 has a library.properties with version set to 2015.9.4:

https://github.com/miguelbalboa/rfid/blob/v1.1.2/library.properties

if I add the library as is the release 1.1.2 will be published on top of the other as the latest version "2015.9.4" that is probably not what you want. To solve this you can remove the tag "1.1.2" or add another commit to fix the release number (basically branching from 7e9a17d and moving the tag 1.1.2 from 7e9a17d to the new commit).

Rotzbua, 2016-01-15:
@cmaglie 
I reported to the repo owner and now it seems fixed :)
Is there any tool to check a library/tags?

cmaglie, 2016-01-15:
great, thanks!

> Is there any tool to check a library/tags?

Not yet, this should be still done manually, we are working on some web service to do so, but not soon.

closed 2016-01-19 by cmaglie