arduino/Arduino #4384: Handling a legitimate .exe file for Library Manager

devinconley, 2016-01-04:
Quick question:
The Arduino library I am writing requires a corresponding listener application on the computer and obviously for Windows this would be a .exe file. If I include it in the "extras" folder, will the entire repo still get flagged for exclusion? If so, is there any recommended way for handling this problem?

Thanks in advance.

cmaglie, 2016-03-04:
Hi @devinconley 

adding an .exe is not permitted for secuiry reasons (in addition we also do an anti-virus check with `clamav`).

By the way it seems a bit odd to store the app inside the library.
Maybe an URL that link to the app in the examples may be sufficient?

devinconley, 2016-03-07:
Sorry, I actually resorted to a linked download in the README a while ago. Should have killed this issue then. 

Thanks.

closed 2016-03-07 by devinconley