arduino/Arduino #4387: Git integration inside the IDE (commit only)

naevtamarkus, 2016-01-05:
I would like to ask for a small feature request: it would be awesome to have small support for Git inside the Arduino IDE.

I understand GIT is really feature-rich and making the IDE fully compliant will be a full project on its own... but having a simple "Commit changes to this file" option in the File menu (with a key shortcut) if the file belongs to a git repo would be AWESOME already. It does not need to do much: just check that git is installed, check if the opened file is part of an existing local repository, and give the user a dialog with a 1-line commit message and OK/Cancel buttons if you click on it (or push the right key combination).

I think this will cover 90% of the use cases with Git, that is already a must for any-sized development projects. Having to do 'git commit -m blabla filename' on a separate console is quite some burden that makes you(me) not use it at the end.

Thanks for a great IDE!

lmihalkovic, 2016-01-06:
that one sounds like a good idea...

cmaglie, 2016-04-28:
I disagree, for me this covers 1% of the git usage, "commit" is just the very last action you do, most of the time you do diffs, add in staging files selectively (not altogether) or even part of a file. Less often you look the log, push/pull, merge or rebase.

This issue inspired #4865, #4816 and #4704 but using them leaves the feeling that they are incomplete, in particular you don't know what you're actually committing (this lead the author of #4865 to implement also the "Diff" command even if not explicitly request but it's really a draft and needs more work). Also having just 1 line for the commit message induce a bad habit, because commits should be well commented (possibly with more than 1 line if needed).

naevtamarkus, 2016-04-29:
I disagree, but I believe it's about perspective.
  - For an mid-expert developer that works with branches to code different features with other people that's absolutely what you should be doing with git, moving back and forth and diffing.
  - For an Arduino dummy like me who works on a single file on his own you use git with commits as a safety net for backtracking and eventually push something to github

Anyway... this is the magic with open source, you never know where a new door may lead you! I honestly think it would be a mistake to put a lid to contributions like those ones because "they are not complete". There is a lot of energy there! Featuring git in an IDE will never be a complete task (or else, it will never seem complete for an expert, otherwise it's a titanic task), but please bear in mind those Arduino newbies like me :-)

But I also understand the risks: if you are going on a road-trip, even if you don't know where are you really going, better make sure the car is in good condition and full of gas.

closed 2021-04-06 by per1234

feraldruid22, 2021-10-16:
Why is this wontfix. It's the greatest idea ever just got my vm crashed losing a days work of Ardu code again yet if there wouldve been an autocommit feature to git built into the ide this would never happened.