arduino/Arduino #4388: IPAddress conversion to uint32_t has wrong endianness

jhannemann, 2016-01-05:
Running the following code

    // 0x0A000102 is the hex representation of 10.0.1.2
    Serial.println(0x0A000102);
    Serial.println(IPAddress(0x0A000102UL));
    Serial.println((uint32_t)IPAddress(2,1,0,10));
    Serial.println((uint32_t)IPAddress(10,0,1,2));  

will output this:

    167772418
    2.1.0.10
    167772418
    33619978

This is because the constructor stores the first octet (most significant byte) at the lowest address of the `_address` union:

    IPAddress::IPAddress(uint8_t first_octet, uint8_t second_octet, uint8_t third_octet, uint8_t fourth_octet)
    {
        _address.bytes[0] = first_octet;
        _address.bytes[1] = second_octet;
        _address.bytes[2] = third_octet;
        _address.bytes[3] = fourth_octet;
    }

This is big-endian behavior. The Atmega is little-endian, so the conversion to `uint32_t`, which is simply done like this:

    operator uint32_t() const { return _address.dword; };

yields the octets in the reverse order, breaking code that interfaces with libraries such as low-level sockets or Adafruit's CC3000 library expecting IP addresses in `uint32_t`.

The easy fix would be to simply reverse the assignment order in the IPAddress constructor, but that may break existing code as the internal data structure is barely hidden. Another fix would be to perform the order reversal in the `operator uint32_t()`.

matthijskooijman, 2016-01-06:
Using a big-endian `uint32_t` representation for IP addresses would probably have been a good idea, if this was done from the outset.

I suspect that changing either the result of `operator uint32_t()`, or the way the constructors or `operator=()` interpret their arguments now would indeed break some code, which I think we should be avoiding. Perhaps adding `toBigEndian()` and `fromBigEndian()` methods or something like those could be considered instead? Alternatively, any libraries or sketches that care about this (like CC3000 you mention) should make sure to convert between the right endiannesses (this could use `htonl()`, which converst from host endian-ness to network (big) endiannes).

john3exonets, 2017-03-08:
For those who may stumble upon this looking for a work around:   Just use a builtin AVR compiler function to swap the byte order:

Serial.println(__builtin_bswap32(IPAddress(2,1,0,10));
// Output:  10.0.1.2