arduino/Arduino #4389: home_and_end_beginning_end_of_doc=false doesn't affect shift+home and shift+end

beargun, 2016-01-05:
The `preferences.txt` setting `home_and_end_beginning_end_of_doc=false` doesn't affect shift+home and shift+end.

`end` will go to the end of the current line, like expected. However also pressing the shift modifier doesn't select everything from the current cursor to the end of the line, but to the end of the document. Idem for `home`.

matthijskooijman, 2016-01-06:
I just tested this on Linux, and indeed it works as expected there (though the preference referenced does not actually change *anything* there, which is a bit unexpected in itself). Looking at the code, [the relevant bit of code is here](https://github.com/arduino/Arduino/blob/master/app/src/processing/app/syntax/SketchTextAreaDefaultInputMap.java#L63-L66). The defaults from RSyntaxTextArea that are being overriden here [can be found here](https://github.com/bobbylight/RSyntaxTextArea/blob/master/src/main/java/org/fife/ui/rtextarea/RTADefaultInputMap.java#L58-L65)

So basically, by default in RSTA home/end go to the beginning of the document on OSX and the beginning of the line everywhere else. Arduino overrides this (with the default home_and_end_beginning_end_of_doc=false`) and lets OSX also go to the begining of the line (but allows switching back to the original behaviour by setting that preference to `true`). This Arduino-override forgets to also override shift+home/end. which is the subject of this issue.

I suspect that this would be improved by:
 - Also remapping shift+home/end in the Arduino override
 - Moving the  Arduino override code out of the OSX if, so it applies to all platforms equally
 - Add an explicit else, to map (shift+)home/end to jump to the begin/end of the document when the preference is true.

This will mean that Arduino has the same behaviour across all platforms. Alternatively, we could decide to stick to OSX conventions and have different defaults on macosx, but even then it would be good to make this difference in the default preferences, not the code I think. Also note that this used to be the case, but it was changed by #3715 to default to begin/end of line on all platforms.

aberope, 2016-08-28:
It happens exactly the same to me.

nobias, 2017-09-21:
+1. This inconsistency is quite annoying...

tomstorey, 2018-04-12:
Is there any movement on fixing this? It is very irritating, to the point I am looking at other IDEs to do my development.