arduino/Arduino #4390: esp8266.github.io contains outdated OTA instructions

julianh2o, 2016-01-05:
As of posting this, this URL still contains information that suggests Arduino-PR-4107-BUILD-421 as a requirement for use of ArduinoOTA.
http://esp8266.github.io/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html#arduinoota-configuration

This page, in contrast, has been updated since the pull was merged and accepted into the release for 1.6.7
https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md

I've just successfully used the BasicOTA example using the second set of instructions with an unmodified version of Arduino 1.6.7 (mac), and the 2.0.0 version of the ESP8266 Arduino boards.

I'd also vote that we promote the "ArduinoOTA" method to being the "preferred" method sooner rather than later to encourage new developers to test it and report any troubles they run into.

cmaglie, 2016-01-06:
@julianh2o 
the OTA patch for the IDE is a community contribution and Arduino is not the direct maintainer of the the documentation you linked.

I guess that the correct place to open the issue is here: 
https://github.com/esp8266/Arduino/issues

/cc @me-no-dev

closed 2016-01-06 by cmaglie

julianh2o, 2016-01-06:
Thanks, will do.