arduino/Arduino #4391: Member functions are not inline automatically

avandalen1, 2016-01-06:
In older Arduino versions, member functions were always inline (this is standard c++). that's not the case anymore.

I use Windows XP
ld.exe of version 1.0.5 (else it crash under Windows XP)
Arduino 1.6.6

matthijskooijman, 2016-01-06:
I don't really think that inlining member functions is guaranteed in any way by C++. In practice, small functions whose definition is available will often be inlined, but in the end it's up to the compiler to decide wether this should or should not happen. If you really want them inlined, I think you can add the `always_inline` function attribute. Alternatively, you could try running gcc directly and add appropriate debugging output options to find out why gcc is deciding not to inline your functions (not sure what those options are, though).

In any case, this issue isn't really something that Arduino can fix, I believe, so I'm closing this issue. You might want to try a newer compiler version, and/or reduce the problem to a small testcase and report a bug with gcc. If you think there *is* something Arduino could change (e.g. call the compiler differently), feel free to leave a comment and we can consider reopening this issue.

closed 2016-01-06 by matthijskooijman

avandalen1, 2016-01-06:
I learned that member functions are always inline in standard c++:
http://www.studytonight.com/cpp/inline-functions.php
But I can use "always_inline" now, as you said, this will help me, thank you.

matthijskooijman, 2016-01-06:
@avandalen1, actually, the `inline` keyword doesn't actually tell the compiler to inline a function, but it changes the linkage of the function to allow it to be defined in a header file (inline linkage prevents the linker from complaining about multiple definitions for the function, which it would otherwise do for functions in header files). For member functions declared in the `class { ... }` block, inline linkage is used automatically, but this doesn't actually mean that the compiler *will* inline those functions.

avandalen1, 2016-01-06:
Thank you. It's more complicated than I thought, it tortures my brain a bit, you may have a link to more information?

matthijskooijman, 2016-01-07:
I don't have anything off-hand. Googling around suggests https://gcc.gnu.org/onlinedocs/gcc-5.3.0/gcc/Inline.html#Inline or http://stackoverflow.com/a/16386407/740048

Reading the first makes me wonder if my earlier comment is entirely correct. As you say, it's complicated, so I might have gotten some details wrongly...

avandalen1, 2016-01-11:
Thank you for the links!