arduino/Arduino #4396: Some libraries versions don't appear in different platforms and arduino versions

MustafaFayez, 2016-01-06:
I tried to install OneSheeld library on arduino IDE on both Windows and macOS with different arduino versions, here is what I get:
1-On Windows 7-X64, with arduino version 1.6.5: all versions of OneSheeld lib appear normally as shown:
[image]

On Windows 7-X64, with arduino version 1.6.7: I can't get the last version of OneSheeld library by using manage libraries in arduino IDE, I only get OneSheeld versions 1.6.0 and 1.5.0 despite it have versions up to 1.7.0. Here are screen shots:

[image]

2- On Windows 10-X86, with both arduino versions 1.6.5 and 1.6.7: all versions of OneSheeld lib appear normally as shown:
[image]

[image]

3-On MacOS ElCapitan, with both arduino versions 1.6.5 and 1.6.7 all versions of OneSheeld lib appear normally as shown: 

[image]

[image]

note: the link of OneSheeld library:
https://github.com/Integreight/1Sheeld-Arduino-Library

iSsO, 2016-01-07:
Yes, I have the exact same problem. The versions of the libraries aren't consistent across different Arduino IDE versions / Platforms.

cmaglie, 2016-01-07:
I tried with a fresh install of 1.6.7 on windows 7 x64, and I see all the versions:

[image]

it seems that your 1.6.7 is reading an outdated library index, maybe you're affected by this bug: 
https://github.com/arduino/Arduino/pull/4361

may you try the "Hourly build" and see if that solves your issue?

MustafaFayez, 2016-01-11:
yeah, it worked with the hourly build, thanks.

alieslam, 2016-01-27:
Hi @cmaglie, 
I am still having the issue and can't find the latest updated library of "1Sheeld 1.8.0" using "Manage libraries" although my friend has the same Arduino IDE version "1.6.7" but for MAC and the library version is updated. 
I have attached a screen shot bellow which shows that i have already installed the pre version 1.7.0 and the latest is not available on the list even there's a version 1.6.1 which is not included in the list.

[image]

sandeepmistry, 2016-01-27:
@alieslam can you please try the latest hourly build? https://www.arduino.cc/en/Main/Software

This has been resolved by https://github.com/arduino/Arduino/pull/4361, however this fix is not in the 1.6.7 release.

Alternatively, you can also workaround for now by deleting ```%LOCALAPPDATA%\Arduino15\library_index.json.tmp.gz``` and then restarting the IDE.

I'm closing this as it is a duplicate of #4272.

closed 2016-01-27 by sandeepmistry