arduino/Arduino #4397: recipe execute fail if there is a space character in windows user name

williamlai, 2016-01-07:
In platform.txt I add recipe for build my own image:

    recipe.objcopy.hex.cmd=my_img.bat
    recipe.objcopy.hex.pattern="{runtime.tools.mytool.path}/{recipe.objcopy.hex.cmd}" "{build.path}\application.axf"

In a user name "foo bar" under windows, the whole command line like this when verbose enabled:

    "C:\Users\foo bar\AppData\Local\Arduino15\packages\arduino\tools\mytool\1.0.2\my_img.bat" "C:\Users\foo bar\AppData\Local\Temp\buildfc17b6afbe0bd59b0e46992e10640eb5.tmp\application.axf"

And it execute fail with message:

    'C:\Users\foo' is not internal or external command

So I trace source code of arduino-core, and find this code is specious in method exec() in Compiler.java 

    CommandLine commandLine = new DoubleQuotedArgumentsOnWindowsCommandLine(command[0]);

The class DoubleQuotedArgumentsOnWindowsCommandLine is not actually "double quoted" command[0]. It only passes the argument to super class CommandLine and the contructor of CommandLine also not adds double quotes. So the actually command executed is 

    C:\Users\foo bar\AppData\Local\Arduino15\packages\arduino\tools\mytool\1.0.2\my_img.bat "C:\Users\foo bar\AppData\Local\Temp\buildfc17b6afbe0bd59b0e46992e10640eb5.tmp\application.axf"

So it seems it needs add double quoted somewhere is the compile process.

benlye, 2019-03-15:
I'm still running into this issue in current v1.8.x versions of the IDE.  Is there any chance of it getting addressed?

Ajaykumar415, 2022-10-16:
The issue still persists, hoping they would fix it :/

cfdeangelis, 2022-12-06:
This issue still persists in 2.0.3, kind of ridiculous in 2022

per1234, 2022-12-06:
Now tracked at https://github.com/arduino/arduino-cli/issues/958

closed 2022-12-06 by per1234