arduino/Arduino #4403: Improve status line feedback

lmihalkovic, 2016-01-08:
At the moment the status reports the following information
* line number when there is no selection
* range of line numbers in the form *n-m* when several lines are selected

It would be nice to change this to a more standard **line : character** in order to simplify matching locations with the output from the compiler's error reporting

lmihalkovic, 2016-01-08:
my bad... it seems line number is all that is actually *required* to match errors locations. might still be useful for a **lint** situation to display **line : character**. I am thinking about including the google c++ lint to help newbies with improving code ( @cmaglie / @facchinm you may consider transferring into a python script the contents of the Arduino code guideline that the IDE could execute and show warning about - don't shoot the messenger, this is just a thought )

[image]

lmihalkovic, 2016-01-19:
Due to lack of interest

atuline, 2020-05-13:
It seems that all requests to add 'column' beside the line # in the Arduino IDE point back to this, #4403 and they've all been closed and with no explanation other than 'duplicate'.

Sure, I could use an external editor (and sometime do), but the simplicity of having the compile/upload, the serial monitor and serial plotter all under one roof are pretty compelling. Furthermore, there's a LOT more tutorials on adding libraries and other boards for the native IDE than for other platforms.

With that said, I (and many others) would really like to see the column displayed beside the line # at the bottom of the IDE. Otherwise, please at least explain why it's not going to be there, especially considering that the IDE continues to get updated.

thebigMoose, 2021-01-31:
I couldn't agree MORE!!  We need column number along with line number to align comments and such.  All professional IDE's have that feature.

FlailAway, 2022-05-26:
Add me to that list, but it seems for some crazy esoteric reason, the devs are digging their heels in. It would make a poor editor, just a little less poor, but whadda I know. <<- that's rhetorical folk!

Trying to line up text in LCD displays would be so much easier with that damn column number, but external it is for those.

closed 2025-05-29 by per1234