arduino/Arduino #4406: Ship appdata.xml with Linux builds

probonopd, 2016-01-09:
Please ship [arduino.appdata.xml](https://github.com/arduino/Arduino/blob/master/build/linux/dist/appdata.xml) with the Linux builds provided on https://www.arduino.cc/en/Main/Software

cmaglie, 2016-01-11:
where this file is expected to be? in the root folder of the zip file (together with the `arduino` executable)?

probonopd, 2016-01-11:
According to the [spec](http://people.freedesktop.org/~hughsient/appdata/), it would normally go to `/usr/share/appdata/arduino.appdata.xml` but since Arduino zip files are not packages that are installed into the system by a package manager I think it's not really specified; close to `desktop.template` would probably make sense. The user or some script acting on behalf of the user will then be able to copy it from there and install it in the system.

commit c71004fd (2016-01-12): Added xml appdata to linux distribution

Fix #4406
closed 2016-01-12 by cmaglie