arduino/Arduino #4407: IDE 1.6.7 doesn't find .h files located in subfolders

plinioseniore, 2016-01-10:
Hi,

for [souliss](http://github.com/souliss) we use a subfolder where we collect several .h files, with the new release those are no longer found.

```
C:\Users\Plinio\Documents\Arduino\arduino-1.6.7-windows\arduino-1.6.7\libraries\souliss\examples\ethernet\e01_HelloWorld\e01_HelloWorld.ino:19:70: fatal error: bconf/StandardArduino.h: No such file or directory
```

```C
// Configure the framework
#include "bconf/StandardArduino.h"          // Use a standard Arduino
#include "conf/ethW5100.h"                  // Ethernet through Wiznet W5100
#include "conf/Gateway.h"                   // The main node is the Gateway, we have just one node
#include "conf/Webhook.h"                   // Enable DHCP and DNS
```

Any hint on how get this compiling as was in 1.6.5?

Thanks.

plinioseniore, 2016-01-10:
Looking for previous issues, I've found several issues related like #1004 and #3080 but what is really strange is that those looks fixing something that was working fine for me before o_O

cmaglie, 2016-01-11:
#1004 and #3080 are correlated but IMHO not the cause of the issue.

I think that the cause of your problem is the library-to-library dependency discovery introduced in 1.6.6 (when library A uses library B, you don't need anymore to include both A **and** B, but only A).
To make this happen reliably the builder must process your sketch include lines **one at a time**, when an include is not found in the current build-path the builder try to find a matching library that is added to the build path and the process is repeated. The builder eventually stop at the first failure, in your case the first include in the sketch is:

```
#include "bconf/StandardArduino.h"          // Use a standard Arduino
```

that is not found in any library, so the failure. To workaround this problem you should put first the:

```c++
#include <Souliss.h>
```

so the builder is able to find the souliss library and add it to the build path.
I don't know if this is possible for you, I see in the example comments that `Souliss.h` should be included as last. Another solution may be to add an empty placeholder `Souliss.h`, and move the current `Souliss.h` to, say, `SoulissCore.h` and change the example as follows:

```c++
#include <Souliss.h> // load library

#include "bconf/StandardArduino.h"          // Use a standard Arduino
// other configurations...
#include "SoulissCore.h"
```

plinioseniore, 2016-01-11:
Hi @cmaglie that fix the issue, I've created a blank .h file that drives the IDE in the right direction.

Actually, the Souliss.h shall be after all the other includes just because those one are simply #defines that spread across all the code, so the Souliss.h include itself the whole code from subfolders. As result the whole library is loaded within the sketch itself.

This was the only way with previous Arduino IDE to get a code organized over several subfolders. If you have a smarter way that works with the latest releases, I will appreciate your feedback.

matthijskooijman, 2016-01-12:
@plinioseniore, since IDE 1.5.x, there is a new library format, triggered by the presence of a 'src' folder and a 'library.properties' file. Using that format, all source files inside src (including subfolders) will be included in the compilation. I suspect this would answer your question?

For more info, see [this page](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification) (I'm not sure if there is a more friendly introduction than that page, though).

IT9IPQ, 2016-01-12:
Just to confirm that with the ide 1.6.7 and souliss-7.1-friariello.1 I got the correct (no library error) compilation by creating an empty file SoulissBase.h into a external library (where Souliss.h) and entering in the sketch at the beginning as the first library the line:

` #include <SoulissBase.h> // load patch to the library for IDE 1.6.7`

Thanks for the info

plinioseniore, 2016-01-13:
@matthijskooijman I know about but haven't tried if works for our case. 

Thanks to all for the help.

closed 2016-01-13 by plinioseniore