arduino/Arduino #4408: instructions on Howto page are out of date

afeld, 2016-01-11:
Looking through https://www.arduino.cc/en/Guide/Howto, I found a couple issues:

1) Step 2 says:

> _Mac OS X note:_ After downloading the IDE, run the `macosx_setup.command`. It corrects permission on a few files for use with the serial port and will prompt you for your password. You may need to reboot after running this script.

but according to the [release notes](https://www.arduino.cc/en/Main/ReleaseNotes) from `ARDUINO 0007 - 2006.12.25`:

> Mac versions no longer require running of the macosx_setup.command script.

Seems like that instruction should be removed from the former?

2) Step 3 says to locate the `drivers` folder, but it's not clear where to find that, since the Mac .dmg only includes the Arduino.app. I don't see it within the "package contents", or in the Linux download.

Lastly, I was looking for the documentation files under https://github.com/arduino but couldn't find them...is that part of the site open source? Happy to open a separate issue if that's a broader discussion. Thanks!

agdl, 2016-01-11:
Well, actually I think this page has to be totally removed because it's let's say **ancient**. The information that can be found in this page can be found in secific pages like the getting started guide of the board... @SimonePDA @cmaglie @facchinm what do you think about removing this?

cmaglie, 2016-01-11:
Yes, please.

agdl, 2016-01-11:
Ok I asked the webmaster @mastrolinux to remove it so I'm closing this!

closed 2016-01-11 by agdl