arduino/Arduino #4411: improve support for compiler feedback

lmihalkovic, 2016-01-11:
Aside from hard errors, the compiler is also outputing warnings that can help users improve the quality if their code. It might be useful to 
* display the information in a summarized list form 
* jump from the warnings to the exact code location in the correct editor tab
* display a warning icon next to the line in question in the editor
* make the behavior optional so as not to overwhelm beginner level users

matthijskooijman, 2016-01-15:
Currently, the compiler highlights one line (not sure which one, the first or last error and/or warning). It would indeed be good if this were expanded to highlight *all* errors and warnings and then jump to the *first* one. Using an icon in addition to the line highlight (and probably also using different colors) would help distinguishing between both as well.

> display the information in a summarized list form 

What did you have in mind here? A table-like form like eclipse uses?

> jump from the warnings to the exact code location in the correct editor tab

You mean that clicking an error in the log window jumps to it?

lmihalkovic, 2016-01-15:
Compiler was bailing on first error. I changed it to collect them all. As I explained somewhere it is not a big deal to do that and ensure the output is not "free-floating" from editor to editor when working with several.

* simple list either as a separate console view (not in your face) or via the QuickAccess (less visible)
* yes jump from console out to location (~works)

@mbanzi  In a school/teaching environment, it makes sense to me that the TA/Teacher would be able to have a given script that collects her 'best practices'. Then the student would compile and see what kind of best practices recommendations come out. Being able to have that info in a separate list view from the standard console is just a small extra step. This is a scheme I wanted to test using google's python cpp lint, and one of the items I alluded to but never explained in initial email.

lmihalkovic, 2016-01-15:
```
 It would indeed be good if this were expanded to highlight all errors and warnings and then jump to the first one
```
[image]

Missing warnings. There are so many i think they are best left for manual access rather than coloring each.

lmihalkovic, 2016-01-19:
Still think it would be useful.