arduino/Arduino #4413: library.properties should be documented / recommnended in LibraryTutorial

matthijskooijman, 2016-01-12:
This page seems to be the canonical introduction to writing a library: https://www.arduino.cc/en/Hacking/LibraryTutorial

However, that page does not mention the src/ directory and library.properties at all, so it probably has not been updated in a long while. Since we want to encourage people to include a library.properties file with their library (it should not be an optional extra, every library should ideally have it), that page should be updated with the new library format in mind (with a library.properties example, and probably using the src/ subfolder, but perhaps mention the alternative without src/ as well). It should probably also mention version numbers (not) supporting the new recursive format, just in case people also need to support old 1.0.x versions (IIRC the latest 1.0.x versions also support the new format).

matthijskooijman, 2016-01-12:
The page should probably also mention the library manager and link to https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ and https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

isthatmyname, 2017-01-25:
The example code on the LibraryTutorial page also requires updating - the example code to create an instance "Morse morse(13);" does not work and should be "Morse morse = Morse(13);"

matthijskooijman, 2017-01-25:
> The example code on the LibraryTutorial page also requires updating - the example code to create an instance "Morse morse(13);" does not work and should be "Morse morse = Morse(13);"

Huh? I think both of these are valid, and actually completely equivalent ways of initializing a variable with class type. Did you actually test the code? What is the error message?

smellai, 2019-06-04:
we have a link to specs here too https://www.arduino.cc/en/Create/FAQ#toc11
we could mention the other pages too