arduino/Arduino #4421: Consider unifying all 'preferences' in a single extensible location

lmihalkovic, 2016-01-13:
Owing to original Processing code, preferences are on a single screen in a dialog, with a more recent addition of a networking tab for the network proxy settings. 

This scheme limits the amount of items that can be added (space constraint), how logically the information can be grouped, as well as mostly prevents (where there's a will there's always another way) external contributions to the preferences (for e.g. boards cannot contain their own editable preferences).

The idea here is to 
* brake the main page into groups (e.g. editor / general / compiler )
* transfer the networking info to their own page as-is
* ensure that other pages can be contributed EASILY

A possible extended goal might be to 
* integrate the board manager as a head page where children pages can be contributed by each board distributor
* integrate the library manager as a page

This is a work in progress showing the with a tiny bit of arm-wrestling the existing code can work in both environment (standalone dialog or page inside the preferences)

[image]

matthijskooijman, 2016-01-15:
This looks like a good suggestion to me. Integrating the library and board manager probably makes these easier to find, which seems good to me.

One caveat is that Arduino has always aimed to remain simple (hence fewer preferences), though I believe the current preferences window isn't particular simple anymore and would benefit from some separation.

lmihalkovic, 2016-01-15:
there is alway the conflicting goals of allowing customization and requiring configuration. These are the rules I will keep in mind in my code
* 100% compatible (means it works out of the box just like it does now)
* clear logical grouping
* possible user/expert mode where the visible choices are not the same (like the expert mode in Chrome preferences)

personally I think that more boxes with fewer logically grouped items beat fewer boxes where everything is thrown-in irrespective of meaning.

lmihalkovic, 2016-01-19:
Due to lack of interest