arduino/Arduino #4424: Legit source doesn't compile

OliverSwift, 2016-01-14:
Don't know since which version this occured but a legit source doesn't compile on 1.6.7.

Just try this modified empty sketch:

```
void
setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Causes this error:
```
Arduino : 1.6.7 (Linux), Carte : "Arduino/Genuino Uno"

sketch_jan13a:7: error: two or more data types in declaration of 'loop'
 void loop() {
           ^
sketch_jan13a:2: error: ISO C++ forbids declaration of 'setup' with no type [-fpermissive]
 setup() {
       ^
...
exit status 1
two or more data types in declaration of 'loop'
```

Note that there is a carriage return after void and before setup().
My coding style: I'm used to write returned function type above function name.

Very minor issue.

matthijskooijman, 2016-01-14:
This is a side effect of the way prototypes are now generated - function headers should be on a single line. This might be fixed at some point, but for now it's tricky to get this right with the current approach. One workaround is to manually add a forward declaration for your functions (which, I think, does not need to be on a single line).

OliverSwift, 2016-01-14:
Hey Matthijs,

As mentionned, I consider this minor, but awkward.
I can live with that, IDE is for dummies anyway.

I wanted to report this in case people stumble on this because the error
might sound misleading.
At least users can search issues for that now.

Cheers

 -Olivier

matthijskooijman, 2016-01-14:
Thanks! It's a known issue, but I'm closing this as wontfix for now. If we switch to some other way of generating prototypes, we'll see if we can fix this too.

@cmaglie, was this one on your list of broken examples already?

closed 2016-01-14 by matthijskooijman