arduino/Arduino #4426: Compiles fail after auto conversion from .pde to .ino

KeithV, 2016-01-14:
If you set in Preferences the "Update sketch files to new extension on save" and open a sketch that is a .pde and just do a "save" and then build. You will get an error that the .pde can't be found. I believe that is because the file was a .pde when opened, but when the save occurred the IDE did not switch the build from a .pde extension to a .ino extension and still attempts to compiler the .pde instead of the renamed .ino. Restarting the IDE solves the problem.

lmihalkovic, 2016-01-14:
It has been my experience that the ide does not like renames in general. the first day i tried using it, i changed a file's extension and it took me a while to figure that there were 2 files as the original was still in the build folder 1.6.5 at the time). Thx for the info.

matthijskooijman, 2016-01-14:
@KeithV, could you see if this still occurs with the version shown there: https://github.com/arduino/Arduino/pull/4363#issuecomment-170562023 That pull request contains a lot of cleanups and small improvements, among which some improvements to the renaming code.

facchinm, 2016-08-26:
Fixed by #4363

closed 2016-08-26 by facchinm