arduino/Arduino #4433: Please add the MySQL Connector/Arduino library to Library Manager

ChuckBell, 2016-01-16:
Please add the MySQL Connector/Arduino library to the Library Manager.

This library has been around since 2013 and undergone a number of releases. It was previously hosted on Launchpad but is now on GitHub. 

The latest release is 1.1.0a. It has been rewritten to make it easier to use and easier to write sketches and also to conform to the published guidelines.

The GitHub repository is located at: https://github.com/ChuckBell/MySQL_Connector_Arduino

If there is an issue with the library with respect to adherence to guidelines for the Library Manager, I will be happy to correct them immediately.

Thanks!
Dr. Charles Bell
chuck.bell@oracle.com

cmaglie, 2016-01-19:
Hi @ChuckBell,

there are no tag/releases on your repository https://github.com/ChuckBell/MySQL_Connector_Arduino/releases
you must create one to publish the library in the library manager.

See:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager

ChuckBell, 2016-01-19:
Hi,

I see the tag in github posted 4 days ago. I just ran the command “git push origin <tag>” so maybe that was missing. Please check it again.

Chuck

cmaglie, 2016-01-19:
`gitclones/ChuckBell/MySQL_Connector_Arduino/library.properties: no such file or directory`

almost there :-) the properties file should be named `library.properties` and not `libraries.properties`. So please do the following to fix it:

1) remove the tag `release`
2) rename libraries.properties to library.properties
3) commit and push
4) **after** the above, create a new tag and push

ChuckBell, 2016-01-19:
Hi,

Thanks for the help. I appreciate your patience too. ;) I’ve corrected the file and retagged.

Chuck

cmaglie, 2016-01-19:
> I appreciate your patience too

np, the library is now online.

closed 2016-01-19 by cmaglie