arduino/Arduino #4434: struct does not name a type

vit46, 2016-01-16:
Hi. here an example struct and function in arduino sketch:
struct mytype1 {
	int x;
	double y;
};
mytype1 test() {
	mytype1 a;
	a.x = 1;
	a.y = 1.00;
	return a;
}
Compiler says "error: 'mytype1' does not name a type" (the same thing happens when using typedef too).
There is walkaround to make these work: place user defined types and functions, that used these types into separate .h file and then include them in arduino sketch file. Yes, it works!
But it is not convenient. So I want to know is it a bug or feature? Thank you!

facchinm, 2016-01-18:
Hi @vit46 ,
the function `test()` is automatically forward-declared by arduino-builder, so `mytype1 test()` can appear before `struct mytype1` in the preprocessed sketch (like all the functions of the sketch).
The nightly version of the IDE contains a fix for this issue, can you give it a try?

matthijskooijman, 2016-01-21:
I just tested this with git master and it works as expected. I'll go ahead and close this issue already. @vit46, could you please test the hourly build and confirm it also works for you? If not, we'll reopen this issue and investigate.

closed 2016-01-21 by matthijskooijman

vit46, 2016-01-23:
Thank you guys, yes it's works, I've test it now. Sorry I did not test it earlier. Though it works, but the problem still remains!!! It works in Arduino IDE, but not works in AtmelStudio. AFAIK, AtmelStudio uses Arduino IDE compiler. Please explain me if i not right and help me to solve the problem!

matthijskooijman, 2016-01-24:
> Though it works, but the problem still remains!!

I am not sure what you mean by that?

> AFAIK, AtmelStudio uses Arduino IDE compiler.

AFAIU, Atmel Studio uses the avr-gcc compiler, just like the Arduino IDE does. The Arduino IDE does some extra preprocessing to auto-generated function prototypes, before calling avr-gcc, which (I assume) Atmel Studio does not. However, for the code you pasted, this preprocessing should not be needed (but in some version of the Arduino IDE this preprocessing actually *causes* problems).

vit46, 2016-02-01:
Thank you for explaining. I meant there is no problem in Arduino IDE any more, but I use Atmel Studio Visual Micro plugin where the problem still remains. I assumed the problem in Arduino IDE because Atmel Studio not works without Arduino IDE installed. Now I see the problem may be in Visual Micro plugin itself, therefore I need to contact Visual Micro plugin developers to solve it, right?

matthijskooijman, 2016-02-01:
@vit46, yes, I think you should be contacting the Visual Micro folks indeed. If it turns out the problem *is* in the Arduino IDE after all, just leave a comment with more information and we can reopen this issue.