arduino/Arduino #4439: List of available baud rates in Arduino IDE serial monitor

ranger81, 2016-01-16:
Hi,
This is not a bug but a feature request.
Since I'm developing a lot with the ESP8266 I would like to see 76800 baud rate in the list of available baud rates in the serial monitor. Or even better, please provide a input box to define own baud rates - or in preferences.txt.

Thanks

lmihalkovic, 2016-01-16:
IMHO, looks like a good candidate for something board devs would benefit from being able to override in their packaging.

lmihalkovic, 2016-01-16:
In [PR4102 comment](https://github.com/arduino/Arduino/pull/4102#issuecomment-171289157) i was offering for consideration that rather going down the path of the following implementation:
```
 +      !BaseNoGui.getBoardPreferences().get("serial.disableRTS").equalsIgnoreCase("true"),
 +      !BaseNoGui.getBoardPreferences().get("serial.disableDTR").equalsIgnoreCase("true"));
...
BaseNoGui.getBoardPreferences().get("serial.baudRates")
... // more of the same type of code
```
you might find it advantageous to invest in a more generic way for board packagers to control which values are used where when using their their package (if i recall there was another recent reference to different bd rates values).
```
prefs.get("xxxxx")
```
is IMHO all the ide source code should ever have to care about.

matthijskooijman, 2016-01-17:
This is essentially a duplicate of #4343, so I'm closing this issue. The suggestion of letting board or platform define the list of baudrates (or perhaps define a few *additional* baudrates) does seem new, so I'll copy that one over (since, even if you could type arbitrary baudrates in the dropdown, having a few board-specific shortcuts could be useful).

closed 2016-01-17 by matthijskooijman