arduino/Arduino #4443: New Ring Buffer Library

wizard97, 2016-01-18:
Hi,
I wrote a Ring Buffer library implementation that I would like to share. I have followed the Arduino IDE 1.5 specification.

Here is a link to my repo:
https://github.com/wizard97/ArduinoRingBuffer

Thanks,
Aaron

agdl, 2016-01-18:
@wizard97 if you want to add your library to the library manager please make sure to follow this procedure
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

wizard97, 2016-01-18:
I followed that procedure. 

I conformed to the 1.5 format, and I included a library.properties file. What am I missing?

Thanks in advance,
Aaron

matthijskooijman, 2016-01-19:
The repo and tags look good to me. @agdl, was there anything you were missing in particular?

cmaglie, 2016-01-19:
I added the library, so it's now published on the library manager.

BTW there is a spurious release "v1.2" that points to a `library.properties` containing `1.1`:
https://github.com/wizard97/ArduinoRingBuffer/blob/v1.2/library.properties

as it is, is not a problem for the library manager, but you may want to delete this tag for cleanness and keep only "1.2" and "1.1".

closed 2016-01-19 by cmaglie

wizard97, 2016-01-19:
Thanks for pointing that out, that was an error on my part.

I deleted the "v1.2" tag. Thanks!