arduino/Arduino #4450: Arduino breaks on java version check with java version '9-ea'

aurelijusrozenas, 2016-01-18:
Ubuntu 15.10 if that helps.
```
infro@zveris:~$ arduino
Exception in thread "main" java.lang.ExceptionInInitializerError
	at processing.app.Preferences.setColor(Preferences.java:851)
	at processing.app.Preferences.init(Preferences.java:273)
	at processing.app.Base.main(Base.java:117)
Caused by: java.lang.NumberFormatException: For input string: "9-e"
	at sun.misc.FloatingDecimal.readJavaFormatString(FloatingDecimal.java:2054)
	at sun.misc.FloatingDecimal.parseFloat(FloatingDecimal.java:122)
	at java.lang.Float.parseFloat(Float.java:453)
	at java.lang.Float.<init>(Float.java:534)
	at processing.core.PApplet.<clinit>(Unknown Source)
	... 3 more
infro@zveris:~$ java -version
java version "9-ea"
Java(TM) SE Runtime Environment (build 9-ea+95-2015-12-02-052152.javare.3968.nc)
Java HotSpot(TM) 64-Bit Server VM (build 9-ea+95-2015-12-02-052152.javare.3968.nc, mixed mode)

```

facchinm, 2016-02-08:
Since Java9 is still in "beta", I strongly suggest you to install Java 8 and run `sudo update-alternatives` selecting the current version (unless you are developing some other project with Java9 specific features).
Once out of beta I believe `java -version` will return `9` without any other "spurious" character

closed 2016-02-08 by facchinm

aurelijusrozenas, 2016-02-08:
While that's true, at least it could give a proper message ant not stack trace.

facchinm, 2016-02-08:
Sure, I'll fix it right now :wink:

facchinm, 2016-02-08:
Mmmmh by taking a closer look it seems that you are using an ancient version of the IDE.
In the newer ones that check isn't there anymore. Can you test using latest version (1.6.7 from https://www.arduino.cc/en/Main/Software) ?

aurelijusrozenas, 2016-02-09:
Yes, my version is ancient, however this is from official ubuntu 15.10 repositories. Tested on 1.6.7 - no problem :)