arduino/Arduino #4453: Change request

mmaslyukov, 2016-01-19:
Hi guys, 

Looks like you have an error in bootloader code and in HardwareSerial.cpp of Arduino core.

Way of calculation of UBBRnX registed is bit wrong

UBRR0L = (uint8_t)(F_CPU/(BAUD_RATE * 16L)-1);
UBRR0H = (F_CPU/(BAUD_RATE * 16L)-1) >> 8;

For example, if F_CPU=16Mhz and BAUD_RATE=115200, then UBRR0L should be ~7.68 which rounds to 8 (by the way same as in a datasheet). But when float casts to int 7.68 becomes 7, which obviously is wrong. It leads to communication error in some configurations

My suggestion is modify upper code like this:
UBRR0L = (uint8_t) (F_CPU * 1.0/16L/BAUD_RATE - 1 + 0.5);
UBRR0H = ((uint16_t) (F_CPU * 1.0 /16L/BAUD_RATE - 1 + 0.5)) >> 8;

This changase will not have any impact on size because it is compile time calculations

matthijskooijman, 2016-01-19:
[HardwareSerial.cpp already does this properly](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp#L117):

    baud_setting = (F_CPU / 8 / baud - 1) / 2;

Here, the -1 is applied before dividing by 2, making effectively -0.5 without having to resort to floating point math (and dividing by 8 and then by 2 effectively gives a divisin by 16 as intended).

[Optiboot does a similar thing](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp#L117):

     UBRRL = (uint8_t)( (F_CPU + BAUD_RATE * 4L) / (BAUD_RATE * 8L) - 1 );

AFAICS, this adds BAUD_RATE * 4 before dividing, which is then divided by BAUD_RATE * 8, which effectively adds 0.5 again for rounding. I think optiboot is used for most boards now, though I'm not entirely sure. The stk500v2 bootloader [also has proper rounding](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/bootloaders/stk500v2/stk500boot.c#L381-L389), but the classic `atmega` bootloader does not have it (I suspect you copied your lines from there?). I didn't check any others.

This could be fixed in the atmega bootloader, but it might be worth considering changing to optiboot instead.

mmaslyukov, 2016-01-19:
Thank you for pointing to a proper bootloader

mmaslyukov, 2016-01-19:
>AFAICS, this adds BAUD_RATE * 4 before dividing, which is then divided by BAUD_RATE * 8, which effectively adds 0.5 again for rounding.

Wait a sec.
You said about addition of 0.5. How is that? Whole equation uses integers. there is no any floats.
   
UBRRL = (uint8_t)( (F_CPU + BAUD_RATE * 4L) / (BAUD_RATE * 8L) - 1 );

Same example: F_CPU = 16Mhz and BAUD_RATE=115200
For check I'm using a standard windows calculator, in "Programmer mode" (it uses only integer calculations) 
After calculations UBRRL = 16
Comparing to float calculations UBRRL=16.86 which should rounds to 17

Please correct me if I'm wrong.

matthijskooijman, 2016-01-19:
> Comparing to float calculations UBRRL=16.86 which should rounds to 17
Where did you  get this 16.86 from? I get 16.36:
```
>>> (16000000 ) / (115200 * 8) - 1
16.36111111111111
```

> You said about addition of 0.5. How is that? Whole equation uses integers. there is no any floats.

Consider this formula:

(F_CPU + BAUD_RATE * 4L) / (BAUD_RATE * 8L) - 1 )

which is (assuming floating point calculations) equivalent to:

(F_CPU / (BAUD_RATE * 8L) + (BAUD_RATE * 4L) / (BAUD_RATE * 8L) - 1 )

which simplifies to:

(F_CPU / (BAUD_RATE * 8L) + 4L / 8L - 1

which is of course:

(F_CPU / (BAUD_RATE * 8L) - 0.5

As a simpler example, suppose you want do x/2 and do proper rounding. With floating point you would do (int)(x / 2 + 0.5), but with integer calculations you can instead write (x+1) / 2 as well, which gives you the same result.

mmaslyukov, 2016-01-19:
> Where did you get this 16.86 from? I get 16.36:

Last formula:
(F_CPU / (BAUD_RATE * 8L) - 0.5

After apply values:
```
>>> 16000000/(115200*8)-0.5
16.86111111111111
```

Now I can see that this formula is correct and do calculations in right way. It already has 0.5 delta
(F_CPU + BAUD_RATE * 4L) / (BAUD_RATE * 8L) - 1 ) = 
= (F_CPU / (BAUD_RATE * 8L) - 0.5 = 
= (F_CPU / (BAUD_RATE * 8L) -1 + 0.5

Thank you for your time.

closed 2016-01-19 by mmaslyukov

matthijskooijman, 2016-01-19:
> Last formula:
> (F_CPU / (BAUD_RATE * 8L) - 0.5

This formula already has the 0.5 correction applied that allows to do truncating instead of proper rounding (so properly rounding this result will give the wrong value). But I think you already figured that out :-)

> Thank you for your time.

You're welcome :-)