arduino/Arduino #4454: Arduino IDE misbehaves with links (junctions) on Windows

faxik, 2016-01-19:
I develop a large project that is split into libraries. I used to make links (junctions) from development directories to the Arduino Libraries directory. That allows for quick edit-and-test cycles, as the changes are in effect immediately. This worked good in 1.x; this semi-worked in 1.6.5 and seems to be completely broken in 1.6.6 (maybe, somehow related to Java 8 or new library discovery): IDE just doesn't "see" the linked libraries.

Furthermore, uninstaller deletes not the links, but traverses and deletes all the files which can be very disasterous.

facchinm, 2016-01-19:
Leaving here for the discussion about the uninstaller.
Moving to https://github.com/arduino/arduino-builder for junctions not traversed bug

matthijskooijman, 2016-01-19:
@facchinm, Note that this is a duplicate of https://github.com/arduino/arduino-builder/issues/90

facchinm, 2016-01-19:
Right, closing arduino/arduino-builder#98 as duplicate :smile:

facchinm, 2016-02-08:
Completely moved to arduino/arduino-builder#90, closing this one

closed 2016-02-08 by facchinm