arduino/Arduino #4458: This happens every time and i can't understand or fix it for macintosh

andrew91203, 2016-01-21:
Arduino: 1.6.7 (Mac OS X), Board: "Arduino/Genuino Uno"

Sketch uses 2,564 bytes (7%) of program storage space. Maximum is 32,256 bytes.
Global variables use 56 bytes (2%) of dynamic memory, leaving 1,992 bytes for local variables. Maximum is 2,048 bytes.
/Volumes/Macintosh HD/Users/andrewkane1/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Volumes/Macintosh HD/Users/andrewkane1/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -carduino -PCOM1 -b115200 -D -Uflash:w:/var/folders/ws/6nxbt0jx5wbgchbxpj_ctg0h0000gn/T/build397487f7e3b97fe8fb597c0f2febd421.tmp/Sweep.ino.hex:i 

avrdude: Version 6.0.1, compiled on Apr 14 2015 at 16:30:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Volumes/Macintosh HD/Users/andrewkane1/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/AndrewsMac2/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : COM1
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: ser_open(): can't open device "COM1": No such file or directory
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
if any one knows that this means and how to fix it will save me money that i have wasted.
And also i would like to change COM1 for the script editor, and again using a mac

matthijskooijman, 2016-01-21:
> avrdude: ser_open(): can't open device "COM1": No such file or directory

The problem seems to be that you have not selected any port (in tools -> port), so it tries using the default COM1 port (which is valid only on Windows). If the tools->port menu is grayed out for you, the IDE cannot find any serial ports available, so you are likely running into a driver issue (but I'm really not sure how that works on OSX, I was under the impression that the Arduino boards serial drivers worked out of the box). Is this an official board?

In any case, this does not look like an IDE bug to me, but a problem in your setup. For help resolving those, we recommend using the Arduino forums, keeping this bugtracker clean for actual bugs and feature requests. I'm closing this issue (but there can still be comments left if needed).

closed 2016-01-21 by matthijskooijman

darcese, 2016-07-16:
I had the same issue.

I went to tools then port and then selected usbmodem 1421 and it worked

TKaak, 2018-08-28:
you da best! was giving me a headache right off the bat and thats never a fun way to start a new project