arduino/Arduino #4467: Arduino Due - GSM Shield

ManuelGonzalezR, 2016-01-22:
Hello everyone, I want to know if there is someway to use the Arduino Due and the GSM Shield, this question is because I am using the Arduino Mega, for a project that requires more SRAM space (I have tried several ways to variables code optimization), then I see that Arduino Due has 96Kb in SRAM, and is perfect for the project, however it say that is not compatible with GSM shield.

Do you know if is possible to use this shield with Due or if in someday you are thinking to make it compatible.

Other solution may be to increase the SRAM of Arduino Mega, it is possible?

Many thanks for your answsers. Best regards.

q2dg, 2016-01-22:
Sorry for not answering you...I only wanted to remember this: https://github.com/arduino/Arduino/issues/4159

ManuelGonzalezR, 2016-01-22:
Many thanks with that is enough :) 

I don´t know what happend but I can resolve the problem using function with local variables, the counter of global variables goes down and the warning desappear, Many thanks.

closed 2016-01-22 by ManuelGonzalezR

Daniel-dk, 2016-09-05:
the issue is the hard coded and custom software serial implementation for AVR in the Arduino library - it is the reason for incompatibility with the Arduino DUE and Zero ( they are ARM and not AVR based  so the software Serial implementation will not work for them).

The Software Serial  communication and buffers are woven into into the rest of the library in such a way that it is difficult to replace it with a "standard"/ hardware Serial interface ( .read(), .peak() , .print() , .available() ).

a rewrite of the library will be necessary that uses "regular" Stream objects, so one can use SoftSerial or a "real" Serial port ( UART or USART ).