arduino/Arduino #4468: Error in arduino debug with two files

ManuelGonzalezR, 2016-01-22:
I have discovered and interesting error in the debug of arduino, it is when there is two related files in arduino IDE, and you are compiling, the error can be from the two files but arduino IDE don´t say the file where it is origined only say what line is.

I hope it can be usefull for you. Best Regards.

per1234, 2016-01-22:
Would you mind providing a simple example of files that cause this issue and the error please.

ManuelGonzalezR, 2016-01-22:
[ARDUINO1.zip](https://github.com/arduino/Arduino/files/101430/ARDUINO1.zip)

This is the example file, the principal file is Arduino 1, and there is a second file related to it called Arduino2. The Arduino 1 has an error and when you compile it, the debugger send you to the Arduino 2, but with the line of the Arduino 1.

per1234, 2016-01-22:
Thanks, I reported this issue at https://github.com/arduino/arduino-builder/issues/99 a few days ago. Until the bug is fixed the workaround is to either manually declare the function prototypes for any function defined after the first tab, so you would change ARDUINO1.ino to:
```arduino
void hola();  //function prototype for hola()
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
Serial.println("Hello world");

}

void loop() {
  // put your main code here, to run repeatedly:

  //Here I am going to put and error.
a=3;
}
```
or you can use Arduino IDE 1.6.6 which doesn't have this bug. Neither are very good options I know, hopefully the developers will fix this serious issue soon.

ManuelGonzalezR, 2016-01-22:
Ok... Many thanks for your answer.

closed 2016-01-22 by ManuelGonzalezR