arduino/Arduino #4471: SerialUploader NullPointerException: Serial upload broken in 1.6.8

probonopd, 2016-01-23:
Using 1.6.8 Hourly Build 2016/01/19 08:26 and previous, uploading via the serial port seems to be entirely broken and leads to 

```
processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:179)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1187)
	at processing.app.Sketch.exportApplet(Sketch.java:1161)
	at processing.app.Sketch.exportApplet(Sketch.java:1133)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2417)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.NullPointerException
	at java.lang.String.replace(String.java:2240)
	at processing.app.helpers.StringReplacer.replaceFromMapping(StringReplacer.java:97)
	at processing.app.helpers.StringReplacer.replaceFromMapping(StringReplacer.java:89)
	at processing.app.helpers.StringReplacer.formatAndSplit(StringReplacer.java:37)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:174)
	... 6 more
```

I can reproduce this on Linux and OS X.

probonopd, 2016-01-23:
I can reproduce this with __ESP8266__ and also with an __Arduino Pro Mini__, so I assume it is an issue with the Arduino IDE.

probonopd, 2016-01-23:
@me-no-dev points toward https://github.com/me-no-dev/Arduino-1/commit/b3c91ab85b6aa4dddc7e124733e2fa1084b13694#diff-db26f03d5207a3a0d217244f3de0950fR156 and suspects that `boardPort.getPrefs().get("iserial")` returns NULL, and should be changed to `boardPort.getPrefs().getOrExcept("iserial")` to trigger an exception and not fill the item.

me-no-dev, 2016-01-23:
here is a better link: https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/packages/uploaders/SerialUploader.java#L156-L161

probonopd, 2016-01-23:
I can confirm that the change proposed @me-no-dev works for me.

cmaglie, 2016-01-25:
Fixed by #4472

closed 2016-01-25 by cmaglie

dethoter, 2016-10-27:
I still have NullPointerException during upload on the last stable version of SDK.
**OS:** Linux
**Version:** 1.6.12
**Steps to reproduce:** 
```
$ arduino --upload --board arduino:avr:mega:cpu=atmega2560 --port /dev/ttyUSB0 Sketch.ino
```  

What should I do?

facchinm, 2016-10-28:
Hi @dethoter , you are encountering another problem with CLI that we fixed in mainline. If you use the nightly build you won't encounter this bug.

dethoter, 2016-11-01:
@facchinm, thank you, it was very helpful.