arduino/Arduino #4473: Where to find: Temboo-1.1.4.zip

thsturm, 2016-01-23:
On trying to build the Arduino IDE, I cloned Arduino.git and run ant.
All zip-files are downloaded and unzipped, except the Temboo-1.1.4.zip :disappointed: 
```
untar-unzip-download:
      [get] Getting: https://github.com/temboo/Temboo/archive/1.1.4.zip
      [get] To: /Users/thomas/Workspace/GitHub/Arduino/build/Temboo-1.1.4.zip
      [get] Error opening connection java.io.FileNotFoundException: https://github.com/temboo/Temboo/archive/1.1.4.zip
      [get] Error opening connection java.io.FileNotFoundException: https://github.com/temboo/Temboo/archive/1.1.4.zip
      [get] Error opening connection java.io.FileNotFoundException: https://github.com/temboo/Temboo/archive/1.1.4.zip
      [get] Can't get https://github.com/temboo/Temboo/archive/1.1.4.zip to /Users/thomas/Workspace/GitHub/Arduino/build/Temboo-1.1.4.zip
```
Searching GitHub gives no result for "Temboo" from gituser "temboo".

How to build Arduino from a fresh git clone?

probonopd, 2016-01-24:
If you don't really care about Temboo, you can delete the Temboo line in https://github.com/arduino/Arduino/blob/19a9e4d5e046e79b21ebbad12e97fdc8c05aba54/build/build.xml#L203

thsturm, 2016-01-24:
Yes, I know this 'workaround'.
But, if there is no 'Temboo' archive in GitHub, why not drop this line in the 'official' Arduino repository?
Or, if there is another place in the internet, let this line point to the right place?

cmaglie, 2016-01-26:
https://github.com/temboo/Temboo used to be the official Temboo repository, but they probably moved it elsewhere?

@cormacdriver-temboo @temboo
may you explain what's happened?

cmaglie, 2016-01-27:
ok we sorted this out, Temboo library is available again here:

https://github.com/arduino-libraries/Temboo

(the old link still work and redirects to the new one, so the build should work again).

closed 2016-01-27 by cmaglie

thsturm, 2016-01-27:
BUILD SUCCESSFUL
Total time: 18 seconds

Thank you, it works!