arduino/Arduino #4474: New library Request - Advanced PS2 keyboard Support

techpaul, 2016-01-23:
Arduino PS2 Keyboard FULL keyboard protocol support and full keys to integer coding

Suite of libraries for differing levels of requirements
First Public Release Version V1.0.2

Smaller footprint for data and code, than any others found. Tables kept in Flash

For other versions that convert the up to 8 byte keycode sequences to a single integer for all keyboard types or allow you to get the full UTF-8 configurable for any LATIN keyboard with extra functions keys as simple codes see associated reporitories for PS2KeyAdvanced and PS2KeyMap libraries. Yes these libraries even provide codes for multimedia keys and Function keys F1 to F24 (yes that is F24).

Platforms
    Arduino AVR - tested on Uno and Mega 2560
    Arduino SAM - tested on DUE

Test Environment
    All platforms Hardware version R3
    Arduino IDE V1.6.7
    DUE Board Manager package V1.6.6

Each library has files with LOTS of comments in, probably more comments than code.

extras folder for each has extra documentation and images of what happens on the wires

examples folders for each has at least one example ranging from keyboard to serial output, to other
functions to control the functions even LEDs, repeats and which country mapping, all way to using
to display on LCD and using arrow keys, HOME, END etc to move cursor.
    
-PS2KeyRaw - https://github.com/techpaul/PS2KeyRaw.git 
   * Basic handle keyboard and return bytes as received from keyboard from 1 byte to 8 byte sequence

-PS2KeyAdvanced -  https://github.com/techpaul/PS2KeyAdvanced.git 
   * FULL PS2 protocol support of ANY key

-PS2KeyMap - https://github.com/techpaul/PS2KeyMap.git 
     * extension to PS2KeyAdvanced to convert integers to allow any keyboard mapping
     * Already included - US, UK/GB, DE, FR
     * Also converts codes within UTF-8 range to their UTF-8 code.

agdl, 2016-01-26:
Hi, if you want t add these libraries to the library manager you have to define a tag. Please follow this guide:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

techpaul, 2016-01-30:
@agdl Thought I had been back through and checked and ensured the tag is available on all 1.0.2

agdl, 2016-02-01:
Done thank you!

closed 2016-02-01 by agdl