arduino/Arduino #4475: Renamed GitHub libraries broken in library manager with CRC match failure error.

tdicola, 2016-01-23:
Apologies for the issue but we were renaming a few Adafruit libraries on github, like http://github.com/adafruit/LiquidCrystal to http://github.com/adafruit/Adafruit_LiquidCrystal, and noticed they unfortunately are now in a bad state and can't be downloaded with the library manager.  Attempting to download and install them fails with the error "CRC doesn't match, file is corrupted" (using 1.6.7):
[image]

GitHub does a redirect for accesses on the old named path, but it looks like something in the library manager might be depending on the URL in the CRC file generation.  Can the following libraries be 're-added' (their old entry removed, then new one added with their new URL) to the library manager so they're back in a good state and using the new URLs?
- https://github.com/adafruit/Adafruit_LED_Backpack (old name was https://github.com/adafruit/Adafruit-LED-Backpack-Library)
- http://github.com/adafruit/Adafruit_LiquidCrystal (old name was http://github.com/adafruit/LiquidCrystal)
- https://github.com/adafruit/Adafruit_GPS (old name was https://github.com/adafruit/Adafruit-GPS-Library/)
- https://github.com/adafruit/Adafruit_FONA (old name was https://github.com/adafruit/Adafruit_FONA_Library)

Let us know if you have any questions, thanks and apologies again for the issue--unfortunately didn't realize renaming a library would cause a problem.

Adding @ladyada too to check out the issue.

closed 2016-02-01 by agdl

ladyada, 2016-02-01:
@agdl hey just so we know for future renames - does this mean we should still re-submit libraries when we change the names or that url wont affect the CRC as of 1.6.8?

cmaglie, 2016-02-01:
You must resubmit the library (or tell us the name changes as you did here).
How many libraries you need to rename?

ladyada, 2016-02-01:
maybe 10 or 20...ill do them and submit all at once if that makes it easier

cmaglie, 2016-02-02:
Yes, thank you!
A list with "old name" -> "new name", like @tdicola did, is good.