arduino/Arduino #4476: ATMEGA32U, Usb setup packet sent, kill serial communication

Ced2911, 2016-01-24:
Hi i was playing with pyusb and arduino, the new possibilities are awesome !
I have noticied a bug, when sending some usb packets, the usb serial output doesn't response anymore.

see:
https://gist.github.com/Ced2911/4d26117018d0a0afb7e5

NicoHood, 2016-01-24:
1. Which OS are you using and which version of the IDE?
2. Did you use an official board?
3. Did you use a modified bootloader?
4. Does the Led still blink or does the arduino completely hang?
5. 8mhz or 16mhz?
6. USB2 or 3 Host? USB2 or 3 Hub?

Ced2911, 2016-01-24:
Win10 x64
I'm using a arduino micro clone 5v/16mhz is plugged in a usb3 port, i'm using Win10 x64, with arduino 1.67.
The arduino is still runing, the led is still blinking.

NicoHood, 2016-01-24:
You might want to consider using linux ;)

Since it is a clone we do not know which bootloader is running. However it should not matter much. I have never used win10 (and never will), so I cannot say if its windows related.

Ced2911, 2016-01-25:
Hi, this is working on linux :) so it's must be related to the windows usb stack  ?

cmaglie, 2016-01-25:
Hi @Ced2911,

this is so experimental and unusual use-case that it's very hard to tell if it's a bug in the atmega, in the USB stack or in the USB Library, it seems also difficult to reproduce.

In any case this issue tracked doesn't seem the right place to discuss it, probably you want to open a new topic in the Arduino Forum (-> forum.arduino.cc) and see if someone else has already experimented with pyusb (or maybe wants to experiment) and eventually come back here if you can determine that there is a bug in the Arduino core, with some more details.

closed 2016-01-25 by cmaglie