arduino/Arduino #4479: Syntax highlight, 8 and 9 are erroneously highlighted

00alis, 2016-01-25:
[image]

Is this a wanted behavior @agdl? If not, on Arduino Create we have a color (#A59362) for numbers if you want to try it while you look into the theme file

Thanks!

peabo7, 2016-01-25:
Looks like it is treating them as octal digits, which is not quite right.  However, if it were writtten as

  ... , 08, 09, 10};

then the 08 and 09 should be highlighted.

lmihalkovic, 2016-01-26:
If I may say, I am really curious about this one as I see the right colorization in my fork. I'm not sure I ever saw numbers colorized by default though.

matthijskooijman, 2016-01-26:
> Looks like it is treating them as octal digits, which is not quite right. However, if it were writtten as

That would actually suggest it's treating all the other numbrs as octal numbers, except for the 8 and 9 (which are not valid octal numbers, of course).

Also, I can't seem to reproduce this either. Here's the line for anyone wanting to copy-paste:

     const int seqLed [] = {3, 4, 5, 6, 7, 8, 9, 10};

@00alis, what Arduino IDE version are you using? Can you perhaps post the full sketch that causes this (after removing everything that is not needed to show the problem)?

00alis, 2016-01-26:
It does it with any sketch, any core. I am on 1.6.7 and I did not modified my theme.txt
[theme.txt](https://github.com/arduino/Arduino/files/104814/theme.txt)

[image]

I am glad this just happens to me :)
thank you guys!

matthijskooijman, 2016-01-26:
@00alis, often copy-pasting sketches instead of (or in addition to) a screenshot helps, because then people can just copy-paste the sketch to try and reproduce the issue (instead of having to type it). It also helps to be sure that things are *identical*, not just similar. Could you paste your sketch in a comment here?

00alis, 2016-01-26:
absolutely
```
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
 const int seqLed [] = {3, 4, 5, 6, 7, 8, 9, 10};
}
```

matthijskooijman, 2016-01-26:
Still doesn't show the problem for me, all numbers are black. I just downloaded 1.6.7 (linux64) to test. Perhaps it's platform-specific? What OS are you running?

00alis, 2016-01-26:
@facchinm did his magic and found out it is the Tinkerkit Library
```
libraries/TinkerKit/keywords.txt:8	LITERAL1
libraries/TinkerKit/keywords.txt:9	LITERAL1
```

sorry for all the noise <3 and thanks again!

closed 2016-01-26 by 00alis

agdl, 2016-01-26:
@00alis I think you have some library that defines those number as KEYWORDS2 in the keywords.txt file!

agdl, 2016-01-26:
Ok just like I supposed :)