arduino/Arduino #4484: request to add NeoPixelBus library

Makuna, 2016-01-26:
https://github.com/Makuna/NeoPixelBus

I want to include specific branches.

The master branch is generic but old and I really don't want to include it.
The NeoPixelAnimator branch is generic and I would like to include it but it does not support the esp8266 well.

The DmaDriven branch is esp8266 specific and this currently supports the new library format and the one I wish to include.

agdl, 2016-01-26:
Hi, your library is without a tag, so at the moment it can't be added to the library manager. 

Please follow this guide:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Makuna, 2016-01-26:
So is the suggestion that expose releases targeted from branches tagged with specific platforms then?

"v1.0"  - all platforms but has esp8266 support that has issues due to esp SDK changes
"v1.0-Esp8266Uart" - esp8266 specific, Esp01 requires this due to pin exposure
"v1.0-Esp8266Dma" - esp8266 specific, All but Esp01 can and should use

?

cmaglie, 2016-02-10:
the rule is: everything you **tag** is going to be published in the library manager and it will appear with the name and version written in the file `library.properties` corresponding to each tag.

Currently I see no tags in your library: https://github.com/Makuna/NeoPixelBus/releases so as it is now cannot be published.

> "v1.0" - all platforms but has esp8266 support that has issues due to esp SDK changes
> "v1.0-Esp8266Uart" - esp8266 specific, Esp01 requires this due to pin exposure
> "v1.0-Esp8266Dma" - esp8266 specific, All but Esp01 can and should use

that would not work because the IDE will consider all those tags as versions of the **same** library, and the IDE will ask the user to always upgrade to "1.0".

The best solution IMHO would be to have a single library that works in the three cases. I don't know the details but wouldn't thibs be accomplished by using #ifdef with platform values?

Makuna, 2016-02-10:
AVR doesn't include basic STL functionality while some platforms do.  New features in C++ can't be used easily without them (see functional).   The latest designs rely on this to make it easier on the consumer to use these new features.  I have yet to find a STL library that works and includes <functional> for the AVR platform.  Let alone then we have the nightmare of dependent libraries of which you have no solution yet.  Fix this, and I am halfway there.

Further, how does a user set one of these #define that works within a library, as one platform has two versions (esp has uart and dma, due to dma pins not exposed on some modules and the uart pins is valuable to loose for those that do expose the pins).  The only way I have had it work is by modifying the library header file (not a nice thing to require novice users to do and a nightmare to explain and have them manage if they work between two versions, and I do have customers that use both the esp versions).  Again, if Arduino IDE exposed **per sketch** way of seeing and adding defines that gets applied across all files being compiled then I would have a workable solution.  Today those customers use GIT client to select the branch and it just magically switches for them (easy to explain, quick to happen) but still annoying to customers of the library.

I could create two completely different repos, but this creates a maintenance nightmare for me as there is a lot of shared code between the versions.

So, its not so simple.   And I am stuck not being able to expose my library in your system due to a bunch of limitations out of my control.

sandeepmistry, 2016-02-10:
> I could create two completely different repos, but this creates a maintenance nightmare for me as there is a lot of shared code between the versions.

That would be my suggestion, as it sounds like you actually have three separate libraries.

You can try using Git submodules for the shared code between the repos.

Makuna, 2016-02-10:
@sandeepmistry It truly DOES NOT SOUND like I have separate libraries to me.  While I appreciate your feedback, I do not agree with it.  I do want to merge into one code base, but the Arduino IDE limits me in doing so (lack of per sketch define, lack of modern STL library support built in).  

Truly, I would rather not support the library manager feature than require that I have three repos to do so.

Feedback from my customers so far has been ignore the library manager and just use the GIT client to switch, they all think its faster and easier anyway.  Many are switching to PlatformIO due to many reasons so most don't care about library manager anyhow. I don't like using Git client solution, but Arduino IDE has in the past left me without an option and is continuing to provide a solution that doesn't help me.

cmaglie, 2016-02-10:
When I build with the esp core for Arduino (using http://arduino.esp8266.com/stable/package_esp8266com_index.json) I see that the following compile time variables are automatically defined:

```
ARDUINO_ESP8266_ESP0
ARDUINO_ARCH_ESP8266
ESP8266
```

this is the command line used for the build by the IDE (output obtained in verbose mode):

```
     [exec] "/home/cmaglie/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-
gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-
I/home/cmaglie/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/tools/sdk/include" -c 
-Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 
-std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L 
-DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  
-DESP8266 "-I/home/cmaglie/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" 
"-I/home/cmaglie/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/generic" 
"/tmp/build2728d881e67daa8f4a86af4dd14f3c37.tmp/sketch/sketch_feb11a.ino.cpp" -o 
"/tmp/build2728d881e67daa8f4a86af4dd14f3c37.tmp/sketch/sketch_feb11a.ino.cpp.o"`
```

@Makuna 
may you use `ARDUINO_ESP8266_ESP0` to select code for the Uart-based vs Dma-based? something like:

```
#if defined(ARDUINO_ESP8266_ESP0)
   ....uart based esp....
#elif defined(ARDUINO_ARCH_ESP8266)
   ....all other (dma based) esp...
#elif defined(ARDUINO_ARCH_AVR)
   ....avr....
#else
   #error Architecture not supported
#endif
```

> I have yet to find a STL library that works and includes "functional" for the AVR platform.

take a look at this one: https://github.com/kibergus/StandardCplusplus

Makuna, 2016-02-11:
But for other Esp, the user needs to be able to select between DMA and UART, based on which pins they have available in their designs.  This doesn't solve that problem.

Thanks for the pointer to that library, I will look into it.

Makuna, 2016-02-12:
Ok, tried the reference library.  Its a version of STL that is basically OLD!  Its hard for me to tell exactly how old as it looks like bits have been merged in over time, but the functional is not very easy to use due to using older harder to use templates.  The whole reason I use the newer version with my library is to make it easier on the consumer and the referenced one is missing these.

I really wish the Arduino AVR environment would just include it.  Its not really a library as much as part of the environment. It was real nice that the C++ compiler got moved up to a modern version, but the newer STL makes it a lot easier to consume those new features exposed by the new standards.

Makuna, 2016-02-14:
Ok, I ported an STL myself, and it looks like the Arduino precompiler for INO doesn't understand function templates correctly and errors on them and this stops my advanced library being used for AVR.
 in an included header I have the following perfectly good STL/C++...
```
typedef function<void(float progress)> AnimUpdateCallback;
```
that for AVR builds returns the error "error: 'function' does not name a type".

Makuna, 2016-02-29:
UPDATE:  After a complete architecture change, I now have one library and a release was published.  So this is ready to be reviewed and closed.

agdl, 2016-02-29:
Done thank you!

closed 2016-02-29 by agdl

Makuna, 2016-03-02:
How long does this take?  Its been two full days and it still isn't showing up in the LibraryManager.  My other Libraries it happened overnight.