arduino/Arduino #4485: request to add Rtc library

Makuna, 2016-01-26:
https://github.com/Makuna/Rtc

agdl, 2016-01-26:
like for #4484 this library is without a tag

cmaglie, 2016-01-26:
You should also give it a less generic name by adding a prefix, like MakunaRtc.

Makuna, 2016-01-26:
Do you mean change the class name and folder names to be MakunaRtc?  Or just the name in the properties file?

cmaglie, 2016-01-26:
I mean the library name in properties file, because people may search for "Rtc" and find your library, thinking about a generic Rtc library, but confused afterwards by the fact that is a library specific for some models of RTC.

Makuna, 2016-01-26:
Tag added.  "v1.0"
I updated the library title to MakunaRtc as suggested.

agdl, 2016-01-27:
@Makuna please delete the tag v1.0 (a .x is missing) and rename the tag v1.0.0 into 1.0.0 (without the v)  in order to be compliant with the other libraries

Makuna, 2016-01-27:
mmm, you can't require all tags to be just used for your system.  I could have the tag v1.0 for other purposes (other library managers) that you need to ignore/filter.  How do you plan to support this?

agdl, 2016-01-28:
I added it. You should see it in the library manger in an hour. 

For what concern the tags actually you can write whatever you want but for compliance with all the other library I requested you to follow the "standard". What you actually see in the library manager is what is written in the library.properties associated to the tag that for the sake of simplicity is equal to the name of the tag with only numbers like in all the other libraries. I'm telling you this for what you asked in #4484

closed 2016-01-28 by agdl

Makuna, 2016-01-28:
It is showing up with the wrong name.  What are you seeing?

I see MakunaRtc, but the properties files lists "Rtc by Makuna".

agdl, 2016-01-29:
[image]

Makuna, 2016-01-30:
I rebooted my computer the problem went away.  Some sort of caching?  The name I was seeing was what I changed it too before I submitted the fix.  I switched to the name you see just before submitting.  Even restarting the IDE didn't help.  Oh well, problems gone.