arduino/Arduino #4486: Request to add Cytron G15 Shield library

IdrisCytron, 2016-01-26:
The library can be found at the following GitHub link.
https://github.com/CytronTechnologies/Cytron-G15Shield

Thank you.

agdl, 2016-01-26:
Hi you library does not define any tag, so it can't be added at the moment. Please follow this guide:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

IdrisCytron, 2016-01-26:
Hi, I already create a release for this library. Thanks.

agdl, 2016-01-26:
Done! you will see it soon in the library manager!

closed 2016-01-26 by agdl

IdrisCytron, 2016-01-28:
Hi, when this library will available on Arduino Library Manager? Thanks.

agdl, 2016-01-28:
It is already available!

IdrisCytron, 2016-01-28:
I have search "Cytron G15 Shield" on Arduino Library Manager, but nothing found. Please advice.

agdl, 2016-01-28:
[image]

IdrisCytron, 2016-01-28:
My Arduino IDE still can't search for this library. Nevermind, I try to install the latest IDE. Anyway, thanks for your help. :)