arduino/Arduino #4487: Separate the default platform from the IDE project

lmihalkovic, 2016-01-26:
It seems IMHO increasingly clear that this IDE's future is behind it, and that it is just kept on life support with the very minimal (un-avoidable) changes while all the energy is put on Create (and possibly its announced desktop descendent). Consequently It might make sense to separate the builds of the IDE and platform sooner than later.

lmihalkovic, 2016-01-26:
IMHO it would make sense to carve the platforms as a completely separate github repo, allowing the platform issues&requests to live their own lives, especially if Create ends up open sourced (might have no choice if the Create Desktop contains any GPL code): at that point it is likely you'd want to start with an empty issues list for Create.

per1234, 2017-07-04:
Since Arduino AVR Boards is the only platform still in the IDE repository and https://github.com/arduino/Arduino/issues/5976 is much more informative I'm closing this as duplicate.

closed 2017-07-04 by per1234