arduino/Arduino #4488: Improve the search UX

lmihalkovic, 2016-01-26:
Opening the search window covers ~50% of the text we are searching through

[image]

No less inconvenient is the fact that the search window can disappear below the editor. ALT-F makes it come back, but it does not IMHO give the best of impression

[image]

lmihalkovic, 2016-01-27:
I guess something like the following might be somewhat like Arduino Create will do, right?!

[image]

* highlight all matches
* find next/previous to cycle through results
* update highlight/count on each character typed

PickyBiker, 2016-01-27:
Some people might like the search window to appear as part of the main window, others might want it to simply be always on top and moveable, still others might want it to be pinnable to the main window. Lots of options. Personally I would favor any of these over the way it drops below the window now.

lmihalkovic, 2016-01-27:
At the moment ALT-F opens it under the editor. Switching tabs keeps track of each searches. ALT-SHIFT-F opens the same search in a floating window. I thk swtching editor window will close it. But maybe I will just kill the popup. Searching also works in binary files (.hex)

lmihalkovic, 2016-01-27:
Will also likely change the text field into a drop down with a list of recent searches. Same for replace field.

PickyBiker, 2016-01-27:
That would be great!

per1234, 2017-07-04:
I'm normally not a fan of UI transparency but Notepad++ has a transparency option for the find window and I do find it to be an improvement.