arduino/Arduino #4497: IDE splitted sketch, misleading error report

nikiob, 2016-01-27:
### Description:
an error generated in the main tab is reported as an error in last open tab:
### Error report
*(Please observe italic part: "W_I_P_Functions:9", it doesn't even exist, "9" is the wrong line in the main tab)*

Arduino: 1.6.7 (Windows 7), Board: "Arduino/Genuino Uno"

WARNING: Category '' in library OneWire is not valid. Setting to 'Uncategorized'
F:\Elettronica\ArduinoPrgs\sketch_jan27b\W_I_P_Functions.ino: In function 'void setup()':

*W_I_P_Functions:9*: error: 'Error' was not declared in this scope

exit status 1
'Error' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
### View
##### Main Tab
actually the error is line 9, calling an "Error" variable that doesn't exist,
line 2 is Highlighted instead.
[image]
##### W_I_P Tab
No errors on this tab, BUT
error on line 7 is highlighted,
error on unexistent line 9 is reported, 
[image]

per1234, 2016-01-27:
I have reported this issue a week ago at https://github.com/arduino/arduino-builder/issues/99. Until the bug is fixed the workaround is to either manually declare the function prototypes for any function defined after the first tab, so you would add the lines:
```arduino
void someFunction();
void here(byte INvar);
```
to the top of the sketch_jan27b tab or you can use Arduino IDE 1.6.6 which doesn't have this bug.

nikiob, 2016-01-27:
Thank you, your workaround works!
If my understanding of post labels is correct you should mark your post with  "Component: IDE user interface" and "Type: Bug"...
I'd have done the same but I don't know how to do that... sorry.

per1234, 2016-01-27:
Only project owners/collaborators can add labels so we can't do it but I think the issue certainly qualifies for a bug label.

matthijskooijman, 2016-01-28:
@nikiob, the arduino-builder repository uses fewer labels, but I did add a bug label just now. Thanks for the reminder.

I'll go ahead and close this issue now, since it is a duplicate.

closed 2016-01-28 by matthijskooijman