arduino/Arduino #4501: V1.6.7 Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude assumes it's being run by an admin user/install owner

jhowe-sgs, 2016-01-28:
As a workaround for Gatekeeper under Mac OS X, the IDE ships with a simple wrapper script to change the exec permissions of Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude_bin after install. #3468 

However, this script will fail for non-admin users running the Arduino IDE ( students in a lab ) since it doesn't have admin access. #4270 

When pushing code, the faculty members ( local admins ) also get a chmod warning: #4270 

chmod: Unable to change file mode on /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude_bin: Operation not permitted
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude_bin: line 6: ... : Permission denied

A more graceful degradation might be to check to see if the current user is the owner of the file being chmod'ed, or provide a command line fix wrapped in sudo. I don't know how wrapping the chmod call in the avrdude script with a sudo call would bubble up the authentication challenge in the IDE UI.

Thanks

blueskyworld, 2016-06-10:
Im getting the same issue with 1.6.9 on MAC os x 10.11.4
i tried to just upload the blink sketch.

chmod: Unable to change file mode on /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude_bin: Operation not permitted
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude_bin: line 6: ... : Permission denied
[image]

I tried to upload to a Tiny circuits board, and to leonardo board. I was able to upload the sketch to a Arduino Zero board though without problems. Ive used both of the board about year ago without problems. It only started after the 1.6.9 and OS update from 10.10.

Any help would be appreciated Thanks

blueskyworld, 2016-06-10:
Also Im the only user on the Mac and logged in as a Admin

cmaglie, 2016-06-16:
This one should be addressed by #5021, the new version of avrdude is build with libusb statically linked so there should be no more need for the workaround.

cmaglie, 2016-07-27:
Should be fixed by #5021 

@jhowe-sgs @blueskyworld 
can you try the IDE 1.6.10 and confirm this?

Please reopen the issue if the problem is not solved.

closed 2016-07-27 by cmaglie

vit0rg, 2016-10-04:
Not fixed. At least with Arduino IDE 1.7.11 + Mac OS 10.12 + Arduino Yun got this:
`/var/folders/hr/gn_tjxhd0536wkycms6y7r340000gp/T/build8557427975286146435.tmp/Blink.cpp.hex:i 
chmod: Unable to change file mode on /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude_bin: Operation not permitted
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude: line 6: /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude_bin: Permission denied
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude: line 6: exec: /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude_bin: cannot execute: Undefined error: 0`

facchinm, 2016-10-05:
IDE 1.7.11 is not this repo but [this](https://github.com/arduino-org/Arduino) one. Plese check again using IDE 1.6.12 you can download from http://www.arduino.cc/en/Main/Software