arduino/Arduino #4504: Standard in and out

krupski, 2016-01-30:
Hi,

Saw your "Improvement of stdio.h functions". Check out my code:

https://github.com/krupski/Stdinout

May be what you're looking for. Also check out this:

https://github.com/krupski/Print

Has many extra features and improvements.

facchinm, 2016-02-04:
Hi @krupski ,
do you want your libraries to be added to the Lib Manager? 
In this case you must follow [this guide](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) to adapt them to the right format

krupski, 2016-02-04:
Hi,

I repackaged the library according to the guidlines. Hopefully it's 
correct. Please
let me know if I need to fix anything.

-- Roger

facchinm, 2016-02-05:
The structure of Stdinout is correct but you should tag a release and then we can include your library. Instead Print is in pre 1.5 format, you should change the folder structure, add a library.properties and so on

krupski, 2016-02-05:
If you're talking about the Print library I have on here, I have no problems releasing it to whoever would like to use it, but it's not in 1.5 format yet. Were you interested in Print also, and if so, do you want me to fix it up to comply with 1.5?

Thanks. -- Roger

per1234, 2017-05-02:
@krupski you still need to create a 1.0.1 release or tag in the Stdinout repository so that it can be added to Library Manager. See:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ
Also, was it your intention to have a Stdinout.h and Stdinout.cpp in both the root and src subfolder of the repository?

I don't believe it is possible to install that Print library using Library Manager. The Print.h in the Arduino core library gets included instead. If I understand correctly those files are meant to replace Print.h and Print.cpp in the Arduino core library.

krupski, 2017-05-02:
Could you provide me with guidelines as to how to properly package library code for all Arduino versions? I've got quite a few that need it.

per1234, 2017-05-02:
The library folder structure specifications are fairly well laid out here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#layout-of-folders-and-files

Basically the 1.5 format folder structure is to have a library.properties file in the root library folder and then put the source files in the `src` subfolder. The only drawback of that folder structure is the library will no longer be compatible with Arduino IDE 1.0.x. If you want your library to be backwards compatible with IDE versions 1.0.x while still working with 1.5.x and newer you should put your source files in the root library folder or the `utility` subfolder.

The library.properties file is not required for the 1.0 library format but it is required for inclusion in Library Manager and even if the library is not in the Library Manager index it will still make the entry in Library Manager shown when the library is installed manually more informative.

The 1.5 library folder structure is not required for inclusion in the Library Manager index.

per1234, 2020-09-25:
Since it's been a few years and the libraries were never made compliant with the requirements for addition to the Arduino Library Manager index, I'm going to go ahead and close this. @krupski if you ever get around to pushing a tag to https://github.com/krupski/Stdinout, feel free to open a new issue requesting it to be added to Library Manager and we'll be sure to take care of you.

closed 2020-09-25 by per1234