arduino/Arduino #4505: OSX: avr.io not found and etc.

Gargo, 2016-01-31:
I downloaded the latest arduino ide and tried to run this example:
http://www.lucidarme.me/?p=5057

TimerOne not found? Okay, downloaded and installed it manually (TimerOne is a library mentioned on the official website and may be included into build).

```In file included from /Users/apple/Documents/Arduino/mpu_bluetooth_test/mpu_bluetooth_test.ino:1:0:
/Users/apple/Documents/Arduino/libraries/TimerOne/TimerOne.h:39:20: fatal error: avr/io.h: No such file or directory
 #include <avr/io.h>
                    ^
compilation terminated.
exit status 1
Ошибка компиляции.
```

W T F?! avr.io is inside the .app file and must be linked automatically. Have you ever checked your ide "releases"??

Gargo, 2016-01-31:
Checked. The same trouble in windows

matthijskooijman, 2016-02-01:
What board do you have selected in the IDE? It sounds like you're not using an AVR-based Arduino (such as the ARM based Due, or the ESP2866 perhaps?), but the TimerOne library only works for AVR-based Arduinos.

Gargo, 2016-02-01:
Wemos R1 D1

facchinm, 2016-02-02:
It looks like your board is based on ESP2866, so you won't have any luck when using avr-only libraries. TimerOne is one of these (https://github.com/PaulStoffregen/TimerOne/blob/master/library.properties#L9)
You could try removing the include and replacing the calls to timer1 with ESP specific calls.
If you need some help, probably the [ESP community](https://github.com/esp8266/Arduino) can give you some!

closed 2016-02-02 by facchinm

Gargo, 2016-02-02:
Thanks. But it seems you are not right because Wemos has ch340g. And there a lot of models based this chip because it is cheap and in all the cases except of software has the same functionality.

matthijskooijman, 2016-02-02:
@Gargo The ch340g is just the USB->serial chip, the ESP2866 is the main microcontroller that you're compiling code for.