arduino/Arduino #4508: CannotUndo Exception

FlyingEagle, 2016-01-31:
Hi,

i got this

javax.swing.undo.CannotUndoException
	at javax.swing.text.GapContent$InsertUndo.undo(GapContent.java:873)
	at javax.swing.undo.CompoundEdit.undo(CompoundEdit.java:63)
	at javax.swing.text.AbstractDocument$DefaultDocumentEvent.undo(AbstractDocument.java:2802)
	at javax.swing.undo.CompoundEdit.undo(CompoundEdit.java:63)
	at org.fife.ui.rtextarea.RUndoManager$RCompoundEdit.undo(RUndoManager.java:251)
	at javax.swing.undo.UndoManager.undoTo(UndoManager.java:335)
	at javax.swing.undo.UndoManager.undo(UndoManager.java:413)
	at org.fife.ui.rtextarea.RUndoManager.undo(RUndoManager.java:138)
	at processing.app.LastUndoableEditAwareUndoManager.undo(LastUndoableEditAwareUndoManager.java:21)
	at org.fife.ui.rtextarea.RTextArea.undoLastAction(RTextArea.java:1613)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.undoLastAction(RSyntaxTextArea.java:3039)
	at processing.app.Editor$UndoAction.actionPerformed(Editor.java:1596)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.menuDragMouseReleased(BasicMenuItemUI.java:943)
	at javax.swing.JMenuItem.fireMenuDragMouseReleased(JMenuItem.java:586)
	at javax.swing.JMenuItem.processMenuDragMouseEvent(JMenuItem.java:483)
	at javax.swing.JMenuItem.processMouseEvent(JMenuItem.java:429)
	at javax.swing.MenuSelectionManager.processMouseEvent(MenuSelectionManager.java:329)
	at javax.swing.plaf.basic.BasicPopupMenuUI$MouseGrabber.eventDispatched(BasicPopupMenuUI.java:870)
	at java.awt.Toolkit$SelectiveAWTEventListener.eventDispatched(Toolkit.java:2425)
	at java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Toolkit.java:2317)
	at java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Toolkit.java:2316)
	at java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Toolkit.java:2316)
	at java.awt.Toolkit.notifyAWTEventListeners(Toolkit.java:2275)
	at java.awt.Component.dispatchEventImpl(Component.java:4787)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
javax.swing.text.BadLocationException: Length must be positive
	at javax.swing.text.AbstractDocument.getText(AbstractDocument.java:808)
	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.getTokenListForLine(RSyntaxDocument.java:419)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.viewToModel(SyntaxView.java:962)
	at javax.swing.plaf.basic.BasicTextUI$RootView.viewToModel(BasicTextUI.java:1552)
	at javax.swing.plaf.basic.BasicTextUI.viewToModel(BasicTextUI.java:1101)
	at javax.swing.plaf.basic.BasicTextUI.viewToModel(BasicTextUI.java:1074)
	at javax.swing.text.JTextComponent.viewToModel(JTextComponent.java:1395)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.viewToToken(RSyntaxTextArea.java:3067)
[... 540 lines cut ...]
	at javax.swing.JComponent._paintImmediately(JComponent.java:5158)
	at javax.swing.JComponent.paintImmediately(JComponent.java:4969)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:831)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:814)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:814)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:789)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:738)
	at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

matthijskooijman, 2016-02-01:
Do you have any idea what you were doing when this occurred? Can you cause it to happen again?

FlyingEagle, 2016-02-01:
yes it happens after some minutes of editing code, it happend mostly the 
whole time.

i re-installed the ide with same effect.

matthijskooijman, 2016-02-02:
What version of the IDE are you using? Could you perhaps try the hourly build (can be downloaded from arduino.cc) to see if it is fixed in the latest git? If that still breaks, perhaps you could try the version linked here: https://github.com/arduino/Arduino/pull/4363#issuecomment-173495190 (the latter contains some small improvements and cleanups which might also affect your problem).

sandeepmistry, 2016-08-24:
I'm closing this due to lack of activity, please re-open with requested info. if you are still interested.

closed 2016-08-24 by sandeepmistry