arduino/Arduino #4509: Cannot read from Serial CDC after re-opening in OS X El-Capitan

waleedshinwari, 2016-02-01:
This is very similar to #4094 . I am using Arduino DUE with OS X running El-Capitan. The OS detects the CDC port properly and I can "cat" out the USB data transmitted by the Arduino through the file /dev/cu.usbmodemxxxx. However, if I break and then re-run the "cat" command, I read nothing (does not give me a resource busy, it just doesn't read anything).

Tried on latest Hourly build 1.6.8 from Arduino website. Also tried the patch to CDC.cpp in #4094

Tried the following:
1) Disconnect and reconnect USB allows me to read data again, even if the Arduino is still powered on through the Programming port and is still running the code continuously.

2) Suspecting OS X driver, I unloaded and re-loaded the following USB kernel extension tree. Did not help:
com.apple.driver.usb.networking
 -> com.apple.driver.usb.cdc
     -> com.apple.driver.usb.cdc.ecm
     -> com.apple.driver.usb.cdc.acm

3) The same thing happens with the Arduino GUI serial monitor: It works the first time, but then closing it and opening it again does not help. I have to either reset the Arduino code or disconnect/reconnect the USB

sandeepmistry, 2016-03-02:
@waleedshinwari please share the sketch you are testing with and also the specific ``cat`` command you used, and what you mean by "break".

Also, please confirm you tried with the patch from PR #4171.

sandeepmistry, 2016-04-13:
I'm closing this due to lack of feedback, please re-open with requested info. if you are still interested.

closed 2016-04-13 by sandeepmistry