arduino/Arduino #4512: Compilation error: conditional #include inside #if

e-moe, 2016-02-01:
Hi guys, I'm trying to implement one class for two different MCU types using conditional #include. 
Is there any way to avoid error and specific files compilation (e.g. compile only a.cpp or b.cpp but not both of them)?

Arduino 1.6.7. Mac OS X El Capitan.

```
sketch/b.cpp.o: In function `I::test()':
sketch/b.cpp:3: multiple definition of `I::test()'
sketch/a.cpp.o:sketch/a.cpp:3: first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling.
```

sketch:
```
#if defined(__AVR_ATtiny85__)
  #include "a.h"
#else
  #include "b.h"
#endif

void setup() {
  I i = I();
  i.test();
}

void loop() {
}
```

a.h:
```
class I
{
  public:
  void test();
};
```

a.cpp
```
#include "a.h"

void I::test() {
  return;
}
```

b.h:
```
class I
{
  public:
  void test();
};
```

b.cpp
```
#include "b.h"

void I::test() {
  return;
}
```

matthijskooijman, 2016-02-02:
Could you try the hourly build (from the downloads section on arduino.cc)? I believe that this issue might have been fixed in git recently.

e-moe, 2016-02-02:
just checked with hourly build (2 February 2016 10:15:23 GMT) - same results:
```
sketch/b.cpp.o: In function `I::test()':
sketch/b.cpp:3: multiple definition of `I::test()'
sketch/a.cpp.o:sketch/a.cpp:3: first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling.
```

matthijskooijman, 2016-02-02:
Ah, now I looked more closely at your example and indeed, this is not expected to work. The problem is that both .cpp files are compiled, regardless of whether the corresponding .h file is included. Arduino or the compiler has no way to match .h files with .cpp files, so all .cpp files in a sketch are always compiled. There is currently no way to specify what files to (not) compile. I can see ways around this:

 - Put the contents of a.cpp and b.cpp in a big #ifdef block. This block should depend on some macro, but since each .cpp file is compiled on its own, the .ino file does not influence them at all. Adding a custom -D option to the compiler to predefine a macro is alsno not supported, so both .cpp files should #include some .h file which you use to decide between the two.
 - Put a.cpp and b.cpp (and the corresponding .h files) each in a separate library. The Arduino IDE does support detecting library dependencies and only includes only the library that you include in the build (and, with the latest hourly builds, this also works properly for #include inside #if).

Since this isn't so much a bug in the IDE, as well as a mismatch in your expectations (and because it won't be easy to change the IDE to meet your expectations in a reliable way), I'll go ahead and close this issue. Feel free to leave more comments if you have questions, though, or if you think there *is* something the IDE should change.

closed 2016-02-02 by matthijskooijman

lmihalkovic, 2016-02-02:
It might not be far-fetched or difficult to add support for arch specific code in the IDE in a manner similar to what was done in the boards packaging: the IDE could with minimum effort support arch specific folders insde the a sketch's root folder and ensure that the contents only gets compiled when the matching hardware is targetted. Possibly something Arduino Create will support?!

mikaelpatel, 2016-02-02:
@e-moe 
Why is there not a processor check in the cpp files? In any case I would structure this as 1) a single header file with processor dependent sections if needed (private or at least protected) to maintain the same interface, 2) single source file with processor dependent sections.

cmaglie, 2016-02-02:
> Possibly something Arduino Create will support?!

The build process is (and will remain) the same for the Java IDE and for Create.

> It might not be far-fetched or difficult to add support for arch specific code in the IDE in a manner similar to what was done in the boards packaging

We tried to add support for the `arch` folder in the past exactly for that, while transitioning from IDE 1.0 to 1.5, but this change has been received very badly by the community, The opposition has been so strong that we were forced to step back and remove it.

e-moe, 2016-02-02:
@mikaelpatel , @matthijskooijman I tried to show minimal example of code to demonstrate errors. Actually I'm trying to fix build errors in https://github.com/digistump/DigistumpArduino/blob/master/digistump-avr/libraries/DigisparkLCD/LiquidCrystal_I2C.cpp:

```
#if defined(__AVR_ATtiny85__) || (__AVR_ATtiny2313__) || (__AVR_ATtiny167__)
#include "TinyWireM.h"      // include this if ATtiny85 or ATtiny2313
#else 
#include <Wire.h>           // original lib include
#endif

...

#if defined (__AVR_ATtiny85__) || (__AVR_ATtiny2313__) || (__AVR_ATtiny167__)
	TinyWireM.begin();             // initialize I2C lib
#else   // original call
	Wire.begin();
#endif

...
```

lmihalkovic, 2016-02-02:
@cmaglie thank you for sharing that part of the history of the platform. If I may ask, I found some past entries referencing the possibility of adding support for Makefile based builds in the IDE. From what I could gather, the discussions seemed centered around libraries. Has there been any similar discussions regarding support Makefile for sketches?! As I shared previously, I found that once the console stream machinery was straightened up, it was straightforward to address the 'floating' compiler messages - they follow the active editor - and even simpler to support Makefiles - basically remove most of the code from Compiler.java (I decided to expose a preference, but it is not required). In light of this apparent ease, I was wondering if there was not another deeper reason for not support Makefile.

lmihalkovic, 2016-02-02:
@e-moe  If I may share: when dealing with multi-environment code, I tend to favor a solution where inside the ```.h``` (public api) switching is done using the publicly available ```#define```, but then turn them into something specific to the library for the conditionals inside the ```.cpp```. This stems from the desire to try to minimize the work of adapting the code to changes in the environment (say when you switch compilers for e.g.) by avoiding having the copy the same, IMHO, longish ```#if defined (__AVR_ATtiny85__) || (__AVR_ATtiny2313__) || (__AVR_ATtiny167__)```
Additionally when I do a search/grep, I can target directly the library's internal ```#define```s, knowing that the results will only point to the library's code. There is really no rational other than to try to save myself work in the future.

e-moe, 2016-02-02:
@lmihalkovic , yes. but I can't change any of these libraries (`Wire` or `TinyWireM`). They are already in use in many different projects and sketches. And because both of them declaring the same thing (e.g `USI_TWI`) I should make changes only in _client_ libraries (in my case `LiquidCrystal_I2C.`) where this variable is used. In this case I ought to import conditionally one dependent libs.

matthijskooijman, 2016-02-02:
@e-moe, if I understand correctly, the original problem you had with the Wire and TinyWireM libraries  selects between two libraries, while your reduced example selects between files in your sketch, making them significantly different. Did you try your original problem with the hourly build?

IIUC the code you pasted is from the .cpp file in the library you linked. However, what does the .ino sketch look like? If you're using the example from that library, you might be including TinyWireM.h from the .ino file unconditionally? Can you perhaps provide the full sketch and the full errors?

e-moe, 2016-02-02:
@matthijskooijman, yep, I tried hourly build on my original problem - same results. Code samples are slightly different but they shares the same idea. Compare my reduced example:

```
#if defined(__AVR_ATtiny85__)
  #include "a.h"
#else
  #include "b.h"
#endif
```

with my original issue:

```
#if defined(__AVR_ATtiny85__) || (__AVR_ATtiny2313__) || (__AVR_ATtiny167__)
#include "TinyWireM.h"      // include this if ATtiny85 or ATtiny2313
#else 
#include <Wire.h>           // original lib include
#endif
```

matthijskooijman, 2016-02-02:
@e-moe, yeah but the primary difference between your example and the real case is that a.cpp and b.cpp are in the sketch? Or did you create two test libraries for them?

Did you see if your .ino file perhaps unconditionally includes either library?

e-moe, 2016-02-02:
@matthijskooijman, you right - my files was in sketch. they are not separate libs. But I also tried to my compile original issue with Wire and TinyWireM libraries with the same error message.

matthijskooijman, 2016-02-02:
@e-moe, okay. But in your original issue with the two libraries, did you have an #include in the .ino file?

e-moe, 2016-02-02:
@matthijskooijman , this is my ino file:

```
#include <TinyWireM.h>                  // I2C Master lib for ATTinys which use USI - comment this out to use with standard arduinos
#include <LiquidCrystal_I2C.h>          // for LCD w/ GPIO MODIFIED for the ATtiny85

#define GPIO_ADDR     0x27             // (PCA8574A A0-A2 @5V) typ. A0-A3 Gnd 0x20 / 0x38 for A - 0x27 is the address of the Digispark LCD modules.

LiquidCrystal_I2C lcd(GPIO_ADDR,16,2);  // set address & 16 chars / 2 lines

void setup(){
  TinyWireM.begin();                    // initialize I2C lib - comment this out to use with standard arduinos
  lcd.init();                           // initialize the lcd 
  lcd.backlight();                      // Print a message to the LCD.
  lcd.print("Digispark!");
}

void loop(){

}
```

matthijskooijman, 2016-02-02:
And this is your problem: 

    #include <TinyWireM.h>                  // I2C Master lib for ATTinys which use USI - comment this out to use with standard arduinos

Your .ino file always includes TinyWireM, even when not compiling for ATtiny, which causes the TinyWireM to be included and conflict with the Wire library included by LiquidCrystal_I2c.h.

You'll have to apply the same trick from the liquid crystal library here, or perhaps just remove the include altogether (since, I think, Arduino 1.6.6 it is no longer necessary for the .ino file to include all libraries needed, they are also detected if a library includes another library).

e-moe, 2016-02-02:
but my .ino is compiles for attiny and I need to use `TinyWireM.begin();`. Is it possible to omit #include directives in this case?

matthijskooijman, 2016-02-02:
@e-moe, I'm assuming this error occurs when you have a non-tiny board selected in the IDE. Or does this also occur when compiling for an attiny board? Perhaps it is one that is not listed in the `#if defined(__AVR_ATtiny85__) || (__AVR_ATtiny2313__) || (__AVR_ATtiny167__)` line?

e-moe, 2016-02-02:
board is selected correctly. moreover I tried to explicitly set `#define __AVR_ATtiny85__` in my .ino. same error. only when I manually comment out `#include <Wire.h>` inside library build is succeeded and sketch working correctly on board.

matthijskooijman, 2016-02-02:
> board is selected correctly.

Please try to be explicit. What is "correctly"? I originally thought your problem was that your sketch did not compile both for ATmega and ATtiny boards, but now I believe your problem might be that the ATtiny build isn't working?

> moreover I tried to explicitly set #define __AVR_ATtiny85__ in my .ino.

That won't help: The library's .cpp file is compiled separately from the .ino file and does not include the .ino file, so nothing you do in the .ino file will affect the compilation of the .cpp file (though it might cause duplicate definitions in the linker stage after compilation). In particular, any #defines in a .ino file (or .cpp file for that matter) are not visible in another .cpp file.

e-moe, 2016-02-02:
sorry for been not specific enough. my board is "Digispark (Default - 16.5mhz)" from https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json. it's built on ATtiny85 (see http://digistump.com/wiki/digispark/tutorials/digispark). I'm trying to compile example from https://github.com/digistump/DigistumpArduino/tree/master/digistump-avr/libraries/DigisparkLCD

matthijskooijman, 2016-02-02:
Ok, so that should define `__AVR_ATtiny85__` I suppose, but it looks like it does not. I do wonder if this is something for the digispark folks to solve, but I'll have another stab at it. Could you run with verbose compilation enabled (in the Arduino IDE preferences) and then past the full build output here?

e-moe, 2016-02-02:
```
/Users/e-moe/Downloads/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware" -hardware "/Users/e-moe/Library/Arduino15/packages" -tools "/Users/e-moe/Downloads/Arduino.app/Contents/Java/tools-builder" -tools "/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/e-moe/Library/Arduino15/packages" -built-in-libraries "/Users/e-moe/Downloads/Arduino.app/Contents/Java/libraries" -libraries "/Users/e-moe/Documents/Arduino/libraries" -fqbn=digistump:avr:digispark-tiny -ide-version=10608 -build-path "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/Users/e-moe/Documents/Arduino/BasicUsage/BasicUsage.ino"
/Users/e-moe/Downloads/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware" -hardware "/Users/e-moe/Library/Arduino15/packages" -tools "/Users/e-moe/Downloads/Arduino.app/Contents/Java/tools-builder" -tools "/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/e-moe/Library/Arduino15/packages" -built-in-libraries "/Users/e-moe/Downloads/Arduino.app/Contents/Java/libraries" -libraries "/Users/e-moe/Documents/Arduino/libraries" -fqbn=digistump:avr:digispark-tiny -ide-version=10608 -build-path "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/Users/e-moe/Documents/Arduino/BasicUsage/BasicUsage.ino"
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
Warning: platform.txt from core 'Digistump AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'Digistump AVR Boards' contains deprecated recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm, automatically converted to recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{archive_file_path}" "-L{build.path}" -lm. Consider upgrading this core.
"/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -DF_CPU=16500000L -DARDUINO=10608 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR         "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/variants/digispark" "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp/sketch/BasicUsage.ino.cpp" -o "/dev/null"
"/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -DF_CPU=16500000L -DARDUINO=10608 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR         "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/variants/digispark" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/TinyWireM" "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp/sketch/BasicUsage.ino.cpp" -o "/dev/null"
"/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -DF_CPU=16500000L -DARDUINO=10608 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR         "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/variants/digispark" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/TinyWireM" "-I/Users/e-moe/Documents/Arduino/libraries/LiquidCrystal_I2C" "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp/sketch/BasicUsage.ino.cpp" -o "/dev/null"
"/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -DF_CPU=16500000L -DARDUINO=10608 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR         "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/variants/digispark" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/TinyWireM" "-I/Users/e-moe/Documents/Arduino/libraries/LiquidCrystal_I2C" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/Wire" "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp/sketch/BasicUsage.ino.cpp" -o "/dev/null"
[... cut ...]
Using library TinyWireM at version 1.0.0 in folder: /Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/TinyWireM 
Using library LiquidCrystal_I2C at version 1.1 in folder: /Users/e-moe/Documents/Arduino/libraries/LiquidCrystal_I2C 
Using library Wire in folder: /Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/Wire (legacy)
exit status 1
Error compiling.
Invalid library found in /Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/Digispark_Examples: /Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/Digispark_Examples

```

matthijskooijman, 2016-02-02:
Ah, I think I see what's happening:

```
Multiple libraries were found for "LiquidCrystal_I2C.h"
 Used: /Users/e-moe/Documents/Arduino/libraries/LiquidCrystal_I2C
 Not used: /Users/e-moe/Documents/Arduino/libraries/arduino_476479
 Not used: /Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/DigisparkLCD
```
You're not actually using the LiquidCrystal_I2C library you think you're using, but a separately installed one. I think the only way to fix that is to remove the other libraries from your `libraries` directory, though.

e-moe, 2016-02-02:
sorry, one more output (removed unexpected libraries in home folder):

```

/Users/e-moe/Downloads/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware" -hardware "/Users/e-moe/Library/Arduino15/packages" -tools "/Users/e-moe/Downloads/Arduino.app/Contents/Java/tools-builder" -tools "/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/e-moe/Library/Arduino15/packages" -built-in-libraries "/Users/e-moe/Downloads/Arduino.app/Contents/Java/libraries" -libraries "/Users/e-moe/Documents/Arduino/libraries" -fqbn=digistump:avr:digispark-tiny -ide-version=10608 -build-path "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/Users/e-moe/Documents/Arduino/BasicUsage/BasicUsage.ino"
/Users/e-moe/Downloads/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware" -hardware "/Users/e-moe/Library/Arduino15/packages" -tools "/Users/e-moe/Downloads/Arduino.app/Contents/Java/tools-builder" -tools "/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/e-moe/Library/Arduino15/packages" -built-in-libraries "/Users/e-moe/Downloads/Arduino.app/Contents/Java/libraries" -libraries "/Users/e-moe/Documents/Arduino/libraries" -fqbn=digistump:avr:digispark-tiny -ide-version=10608 -build-path "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/Users/e-moe/Documents/Arduino/BasicUsage/BasicUsage.ino"
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
Warning: platform.txt from core 'Digistump AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'Digistump AVR Boards' contains deprecated recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm, automatically converted to recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{archive_file_path}" "-L{build.path}" -lm. Consider upgrading this core.
"/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -DF_CPU=16500000L -DARDUINO=10608 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR         "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/variants/digispark" "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp/sketch/BasicUsage.ino.cpp" -o "/dev/null"
"/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -DF_CPU=16500000L -DARDUINO=10608 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR         "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/variants/digispark" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/TinyWireM" "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp/sketch/BasicUsage.ino.cpp" -o "/dev/null"
"/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -DF_CPU=16500000L -DARDUINO=10608 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR         "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/variants/digispark" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/TinyWireM" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/DigisparkLCD" "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp/sketch/BasicUsage.ino.cpp" -o "/dev/null"
"/Users/e-moe/Downloads/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -DF_CPU=16500000L -DARDUINO=10608 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR         "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/variants/digispark" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/TinyWireM" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/DigisparkLCD" "-I/Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/Wire" "/var/folders/wb/s9v7yzln0hnbxk59jxxvtmh80000gn/T/build801b3f612f03aa981995f14bf66be666.tmp/sketch/BasicUsage.ino [... cut ...] irst defined here
collect2: error: ld returned 1 exit status
Using library TinyWireM at version 1.0.0 in folder: /Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/TinyWireM 
Using library DigisparkLCD in folder: /Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/DigisparkLCD (legacy)
Using library Wire in folder: /Users/e-moe/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/Wire (legacy)
exit status 1
Error compiling.

```

matthijskooijman, 2016-02-02:
Ah, so now the error message changed and there is a different problem. And now I do think you've found an actual bug, though it is a problem in arduino-builder, not the Arduino IDE. So, I've opened up a new bug report here: https://github.com/arduino/arduino-builder/issues/106

We'll have to double-check that my deducation is correct, but I think we have enough info to reproduce and fix this now. If there is anything else we need from you, I'll let you know. Thanks for keeping the replies coming :-)

e-moe, 2016-02-02:
welcome :) thanks to you too for helping me and your effort with finding actual bug!

cmaglie, 2016-02-05:
@lmihalkovic 
> I was wondering if there was not another deeper reason for not support Makefile.

no, there is no deeper reason (though we can still discuss the usefulness of such feature, but this will go offtopic here, maybe we can continue on the specific issue).

About the `arch` folder for libraries, there was a painful process that involved discussions with hundred of emails to reach a shared agreement, so I felt the need to warn you before you going into that path, that's it!